felixendres/rgbdslam_v2 #8: Segmentation fault at PCL when compiling with -std=c++11

JoonasMelin, 2015-02-10:
The use of auto type requires compilation with c++11 option(or c++0x), but this results in PCL libraries failing when trying to start rgbd slam. This is the print that gdb gives:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7600a8c in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() () from /usr/lib/libpcl_io.so.1.7

The code works if you replace the auto type with the real type and recompile with out c++11.
The failure is also present in PCL compiled from their GIT repository.
I managed to fix this on my branch from the hydro branch but the increased use of the auto type etc. in the experimental odometry branch makes compilation without c++11 harder.

I am using ubuntu 14.04 with Indigo.

felixendres, 2015-02-10:
Thanks, for the report. I'll make the changes you suggest. If you have them available as a patch, I would be interested. Otherwise, as the error doesn't occur with my setup (12.04 + Hydro), would you be able and willing to test the changes?

JoonasMelin, 2015-02-11:
Thank you for the quick reply. I created an pullrequest for the hydro branch that fixed the compilation on my system.
I would be more than happy to test any changes that could make the PCL work. Do you have any idea what could be causing the PCL to segfault when compiled with the c++11?

JoonasMelin, 2015-02-11:
This is actually related to this: https://github.com/PointCloudLibrary/pcl/issues/619
I am starting to lean to this being an issue with PCL as the rgbdslam crashes before it enters the main() so there is little that can be done to prevent this crash (other than compiling without the c++11)

felixendres, 2015-02-11:
Wouldn't surprise me if that was the issue. I remember an OpenCV version that didn't even compile when c++11 activated, because a syntax error came after an #ifndef <c++11 activated> :-).
I commented your pull request, but there's nothing for you to do. I'll need to test your changes on hydro before merging. I'll hopefully get to that tomorrow.

JoonasMelin, 2015-02-12:
Thank you for this! I will keep searching for the issue from the PCL side of things if I could find something there.

JoonasMelin, 2015-02-13:
To resolve this issue, you need to go a bit deeper. To my knowledge, it is caused by the differing boost implementations when compiling with c++11 and something else. As the PCl is not compiled with -std=c++11 option, it will contain different boost headers compared to rgbdslam. This results in segmentation fault on the initialization of boost.

To correct this, you need to compile PCL from their git repo (this change might be needed https://github.com/PointCloudLibrary/pcl/issues/980).  One way to make PCL compile with the c++11 option is to add this to the CMakeLists.txt: SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")

When you have the PCL compiled, you need to make install, and change CMakeLists.txt on rgbdslam to use these libraries, this can be done by changing the line: find_package(PCL 1.7 REQUIRED COMPONENTS common io) to find_package(PCL 1.8 REQUIRED COMPONENTS common io)

closed 2015-02-13 by JoonasMelin

felixendres, 2016-11-20:
Just add to JoonasMelin's correct diagnosis:
- Also on 16.04 (with ROS kinetic) pcl will cause a SIGSEV on startup if code is built with the c++11 flag
- g2o requires code to be built with c++11
- Since the central class (and others) of rgbdslam has pcl and g2o includes, it cannot be built without changing either g2o or pcl.