felixendres/rgbdslam_v2 #12: why are my image drift went object dynamic ? help me plese .

supamon, 2015-04-30:
I can edit the problem?thank you My research, I use version rgbdslamv2.
picture http://image.ohozaa.com/view2/y9BCYXObWfxP7f6j

kbalisciano, 2015-04-30:
Try moving back from the camera. In the depth map it looks like you have a little double vision due to the limitations of the camera

supamon, 2015-04-30:
I don't understand. I want c++ file edit such as node.cpp.help me pleas. c-club-zaa@hotmail.com

supamon, 2015-07-20:
c -club-zaa@hotmail.com

supamon, 2015-07-20:
c-club-zaa@hotmail.com

felixendres, 2015-07-20:
It seems the person is very near (less than 80cm) to the camera, therefore the visual features found on the person cannot be located in space and will be ignored. This reduces the amount of features available for estimating the motion. If part of the scene is dynamic the dynamic part may then easily "hijack" the motion estimation. Dynamic scenes are only handled well if the majority of the features are static. I usually had good results, even when people moved in the scene, but as I said, there neeed to be enough static features to "anchor" the motion estimates.

closed 2015-11-09 by felixendres