felixendres/rgbdslam_v2 #13: Hi I have a question on RGBD_slam for ros

SookangJun, 2015-08-04:
Hi I have a question on RGBD_slam for ros.

I want to test rgbdslam_v2 on my computer.
However, I am using ubuntu 14.04 for my research..(so, I can't downgrade to 12.04)
so ubuntu 14.04 doesn't support hydro.
Actually I tried to install with this way
-------------------------------------------------
source /opt/ros/hydro/setup.bash
mkdir -p ~/rgbdslam_catkin_ws/src
cd ~/rgbdslam_catkin_ws/src
catkin_init_workspace
cd ~/rgbdslam_catkin_ws/
catkin_make
source devel/setup.bash

#Get RGBDSLAM
cd ~/rgbdslam_catkin_ws/src
wget -q http://github.com/felixendres/rgbdslam_v2/archive/hydro.zip
unzip -q hydro.zip
cd ~/rgbdslam_catkin_ws/

#Install
rosdep update
rosdep install rgbdslam
catkin_make
    catkin_make
------------------------------------------------------but I got this error
 make[1]: *** [rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
-------------------------------------------------------
So, this is my question.
1. Is there any way that I can use rgbdslam_v2 on ros indigo at ubuntu 14.04?

felixendres, 2015-08-17:
Hi,
I have updated rgbdslam v2, there is now an indigo branch on github, 
which I tested with indigo on 14.04.

Let me know whether it works,
Felix

felixendres, 2015-09-19:
Assuming it works, after a month.

closed 2015-09-19 by felixendres

ram651991, 2016-04-06:
Hi, I get the following error while installing rgbdslam in Ros Indigo. Can anyone kindly help out on this issue...???

> ram@ram-ThinkPad-L440:~/rgbdslam_catkin_ws$ catkin_make
Base path: /home/ram/rgbdslam_catkin_ws
Source space: /home/ram/rgbdslam_catkin_ws/src
Build space: /home/ram/rgbdslam_catkin_ws/build
Devel space: /home/ram/rgbdslam_catkin_ws/devel
Install space: /home/ram/rgbdslam_catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ram/rgbdslam_catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ram/rgbdslam_catkin_ws/build"
####
[  0%] [  0%] [  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  8%] [ 13%] Built target rgbdslam_generate_messages_lisp
Built target rgbdslam_generate_messages_cpp
[ 31%] [ 34%] Built target rgbdslam_generate_messages_py
Built target siftgpu_proj
[ 34%] Built target rgbdslam_gencpp
[ 34%] Built target rgbdslam_generate_messages
[ 36%] [ 37%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o
[ 39%] [ 40%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/glviewer.cpp.o
Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/parameter_server.cpp.o
Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/node.cpp.o
virtual memory exhausted: Cannot allocate memory
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/glviewer.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o] Error 4
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

felixendres, 2016-04-10:
I assume this is a duplicate of #29