felixendres/rgbdslam_v2 #16: About using SiftGPU with cuda

flymaxty, 2015-09-19:
Hi,
I found that if I want to use siftgpu with cuda, I have to comment the following line in siftgpu's makefile
`siftgpu_enable_cuda = 0` (line 17)
but the doc did not say that. Is there something wrong for my steps? or you just forget to comment it?  :)

felixendres, 2015-09-19:
That's a bug. I forgot to take this out after changing the test in line 11. Didn't notice, since I don't use cuda. I'll throw the line out, thanks.

felixendres, 2015-09-19:
Ok, fixed in indigo and hydro branches.

flymaxty, 2015-09-20:
Thank you for your quick reply! :)

flymaxty, 2015-09-20:
Oh, there's other problem I forgot to say, in x64 platform, the following line has a little problem:
`simple_find_cuda = $(shell locate /usr/local/cuda/lib/libcudart.so*)`
the libs is in `/usr/local/cuda/lib64/` not `/usr/local/cuda/lib/` in x64 platform.

felixendres, 2015-12-15:
Hi flymaxty. I couldn't find a sensible way to check for cuda. The current way didn't work and the out-of-the-box build fails if cuda isn't installed. I therefore now disabled cuda in siftgpu's makefile. I am afraid this will interfere with your setup, but should be easy to fix for you: just change "siftgpu_enable_cuda = 0" to "... = 1" in line 11 of the mentioned makefile.

closed 2015-12-15 by felixendres