felixendres/rgbdslam_v2 #17: catkin_make error

Ghinda94, 2015-10-28:
When I run catkin_make in catkin_ws I get the following error message:

/home/ubuntu/catkin_ws/src/rgbdslam_v2-indigo/src/main.cpp: In function 'int main(int, char**)':
/home/ubuntu/catkin_ws/src/rgbdslam_v2-indigo/src/main.cpp:134:10: error: 'class QApplication' has no member named 'type'
  if (app.type() == QApplication::GuiClient){
          ^
/home/ubuntu/catkin_ws/src/rgbdslam_v2-indigo/src/main.cpp:134:20: error: 'GuiClient' is not a member of 'QApplication'
  if (app.type() == QApplication::GuiClient){
                    ^
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/main.o] Error 1
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

What can I do to resolve it? Thank you!

felixendres, 2015-10-28:
The problem is, that you have Qt5 instead of Qt4. You can try to replace the line 134 (in main.cpp)

    if (app.type() == QApplication::GuiClient){

with 

    if (ParameterServer::instance()->get<bool>("use_gui")){

but I suspect that this is just the tip of the iceberg.

Ghinda94, 2015-11-04:
Thank you for the answer. I resolved that error but I get another one. What is the best version of Qt for rgbdslam: Qt4 or Qt5?

kbalisciano, 2015-11-04:
Rgbdslam relies on QT4. See their documentation on how to install QT4 along side QT5 and specify your program path

Ghinda94, 2015-11-04:
I have install Qt4. If I run catkin_make I get the following errors:

In file included from /usr/include/GL/gl.h:2059:0,
                 from /usr/include/GL/freeglut_std.h:128,
                 from /usr/include/GL/glut.h:17,
                 from /home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:24:
/usr/include/GL/glext.h:468:19: error: conflicting declaration 'typedef ptrdiff_t GLsizeiptr'
 typedef ptrdiff_t GLsizeiptr;
                   ^
In file included from /usr/include/qt4/QtOpenGL/qgl.h:79:0,
                 from /usr/include/qt4/QtOpenGL/QtOpenGL:5,
                 from /home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:22:
/usr/include/GLES2/gl2.h:69:25: error: 'GLsizeiptr' has a previous declaration as 'typedef khronos_ssize_t GLsizeiptr'
 typedef khronos_ssize_t GLsizeiptr;
                         ^
In file included from /usr/include/GL/gl.h:2059:0,
                 from /usr/include/GL/freeglut_std.h:128,
                 from /usr/include/GL/glut.h:17,
                 from /home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:24:
/usr/include/GL/glext.h:469:19: error: conflicting declaration 'typedef ptrdiff_t GLintptr'
 typedef ptrdiff_t GLintptr;
                   ^
In file included from /usr/include/qt4/QtOpenGL/qgl.h:79:0,
                 from /usr/include/qt4/QtOpenGL/QtOpenGL:5,
                 from /home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:22:
/usr/include/GLES2/gl2.h:70:26: error: 'GLintptr' has a previous declaration as 'typedef khronos_intptr_t GLintptr'
 typedef khronos_intptr_t GLintptr;
                          ^
/home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp: In member function 'void GLViewer::drawOneCloud(int)':
/home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:341:75: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
         glMultMatrixd(static_cast<GLdouble*>( (*cloud_matrices)[i].data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                           ^
/home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp: In member function 'void GLViewer::drawClouds(float)':
/home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:377:64: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
     glMultMatrixd(static_cast<GLdouble*>( viewpoint_tf_.data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                ^
/home/ubuntu/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/glviewer.cpp:416:75: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
         glMultMatrixd(static_cast<GLdouble*>( (*cloud_matrices)[i].data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                           ^
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/glviewer.o] Error 1
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2

kbalisciano, 2015-11-05:
Yea you have to modify GLES2 and some other files. Are you doing this on an
ARM board?

Ghinda94, 2015-11-05:
Yes. I use Nvidia Jetson TK1. Do you know how to install rgbdslam on it?

kbalisciano, 2015-11-05:
You need to modify a ton of stuff. We were going to put out instructions for 12.04 but I'm working on moving to 14.04 now

Ghinda94, 2015-11-05:
Ok. Thank you!

felixendres, 2015-11-09:
If you don't need the 3D visualization within rgbdslam, you can swap glviewer.cpp with glviewer2.cpp (either in CmakeLists.txt, or in file system), which is just an empty shell I made for someone trying to port it to some embedded hardware with GLES.

kbalisciano, 2015-11-09:
I have the fix for that error somewhere. You need to redefine your pointers in your GLES include file

AlysonQ, 2015-11-20:
Hi kbalisciano ,

Would you teach me how to redefine? I have the same issue now.
Thank you!!

AlysonQ, 2015-11-20:
Hi  kbalisciano,

Would you teach me how to redefine it? I follow the log and redefine and it still have some error as below.
I am also use Tegra TK1.
____________________________________________________________________
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer.cpp: In member function 'void GLViewer::drawOneCloud(int)':
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer.cpp:341:75: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
         glMultMatrixd(static_cast<GLdouble*>( (*cloud_matrices)[i].data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                           ^
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer.cpp: In member function 'void GLViewer::drawClouds(float)':
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer.cpp:377:64: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
     glMultMatrixd(static_cast<GLdouble*>( viewpoint_tf_.data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                ^
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer.cpp:416:75: error: invalid static_cast from type 'qreal* {aka float*}' to type 'GLdouble* {aka double*}'
         glMultMatrixd(static_cast<GLdouble*>( (*cloud_matrices)[i].data() ));//works as long as qreal and GLdouble are typedefs to double (might depend on hardware)
                                                                           ^
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/glviewer.o] Error 1
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
______________________________________________________________________________

and if I swap glviewer.cpp with glviewer2.cpp  in cmakelist.txt.
it shows error.

/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer2.cpp: In constructor 'GLViewer::GLViewer(QWidget*)':
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer2.cpp:20:39: error: type 'QWidget' is not a direct base of 'GLViewer'
 GLViewer::GLViewer(QWidget *parent) : QWidget(parent) { }
                                       ^
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/glviewer2.o] Error 1
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2

Thanks!!

kbalisciano, 2015-11-22:
Take a look at my QT4 repo. The instructions aren't all that clear but the source code has been modified slightly. My instructions show a ghost installation, so basically we install QT4 to make all of the correct symbolic links and then copy our build directory over. This isn't completely necessary but I haven't found time to rewrite the installation documentation

kbalisciano, 2015-11-22:
Also @felixendres I would like to let you know that RGBDSLAM works on the ODROID XU3 under 12.04 Hydro and 14.04 Indigo. We only get ~10-15FPS from the Kinect and it is pretty slow if we use the default number of keypoint, etc but with tuning it performs very well for a sub $100 SOC computer.

felixendres, 2015-12-15:
Nice!

closed 2015-12-15 by felixendres

hmalatini, 2017-02-10:
"and if I swap glviewer.cpp with glviewer2.cpp in cmakelist.txt.
it shows error.

/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer2.cpp: In constructor 'GLViewer::GLViewer(QWidget_)':
/home/ubuntu/catkin_ws/src/rgbdslam_v2/src/glviewer2.cpp:20:39: error: type 'QWidget' is not a direct base of 'GLViewer'
GLViewer::GLViewer(QWidget *parent) : QWidget(parent) { }
^
make[2]: *_* [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/glviewer2.o] Error 1
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2"

**You can resolve it if you change GLViewer::GLViewer(QWidget *parent) : QWidget(parent) with:
GLViewer::GLViewer(QWidget *parent) : QGLWidget(parent)
In your case, you have to do it in ./rgbdslam_v2/src/glviewer2.cpp**

chenzhihuan17, 2017-06-12:
@hmalatini Thank you !  But now raises a new issue:

_/rgbdslam_catkin_ws/src/rgbdslam_v2-hydro/src/misc.cpp: In function 'QMatrix4x4 g2o2QMatrix(g2o::SE3Quat)':
/rgbdslam_catkin_ws/src/rgbdslam_v2-hydro/src/misc.cpp:100:52: error: invalid static_cast from type 'Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4> >::Scalar* {aka double*}' to type 'qreal* {aka float*}'
make[2]: *** [rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/src/misc.o] Error 1
make[1]: *** [rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2