felixendres/rgbdslam_v2 #19: siftgpu library error

AlysonQ, 2015-11-06:
Dear sir

I am trying to catkin_make this package on my own Ubuntu14.04.
But here is something error.

[ 73%] Built target rgbdslam_gencpp
Linking CXX executable /home/ros/catkin_ws/devel/lib/rgbdslam/rgbdslam
/usr/bin/ld: cannot find -lsiftgpu
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ros/catkin_ws/devel/lib/rgbdslam/rgbdslam] Error 1
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
ros@ros-VirtualBox:~/catkin_ws$ 

And I find this sentence in README.md,"To build RGBDSLAMv2 go to your catkin workspace and execute "catkin_make". If you get an error about the missing siftgpu library, execute "catkin_make" again."
I don't understand and it doesn't work.

Please give me a hand.
Thank a lot!!
Alyson

kbalisciano, 2015-11-06:
If you don't need sift, edit the cmake file

AlysonQ, 2015-11-06:
thank you 
I edit the camklist.txt and it works.
and if I need sift what should I do to this error?

felixendres, 2015-11-09:
There's a problem with the CMakeLists.txt, calling the makefile of siftgpu sometimes works and sometimes doesn't. So sometimes just re-runing cmake works. If that fails, you can go to external/SiftGPU and run "make" manually. Afterwards the error should be gone.

I spend quite some time debugging this, but unfortunately I still don't know how to fix this properly.

closed 2015-11-17 by AlysonQ

wojiaojiao, 2017-02-14:
i also plese help me

felixendres, 2017-02-15:
Some more details please.