felixendres/rgbdslam_v2 #20: Issues with Boost > 1.48

kbalisciano, 2015-11-07:
[ 36%] Generating src/moc_qt_gui.cxx
Generating src/moc_graph_manager.cxx
Generating src/moc_openni_listener.cxx
Generating src/moc_glviewer.cxx
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [rgbdslam_v2-indigo/src/moc_qtros.cxx] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [rgbdslam_v2-indigo/src/moc_ros_service_ui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [rgbdslam_v2-indigo/src/moc_glviewer.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [rgbdslam_v2-indigo/src/moc_qt_gui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [rgbdslam_v2-indigo/src/moc_graph_manager.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [rgbdslam_v2-indigo/src/moc_openni_listener.cxx] Error 1
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2

This is due to the way Boost handles messaging. 

https://github.com/csete/gqrx/issues/93

This seems to have the issue fixed but I'm confused as to why I'm having this problem and no one else. I will most down a few boost versions and try to reinstall.

felixendres, 2015-11-09:
Sounds like a version-specific thing. Let me know how it works after the update. If you make a pull request with the precompiler statements mentioned in the link, I'll merge it in.

felixendres, 2016-02-01:
Fixed with pull request #21

closed 2016-02-01 by felixendres

uzhaoyang, 2016-10-11:
I still find same issue with my kinectic, boost 1.53, qt4, ubuntu16.04. 
Some help from you will be grateful.
thx.

Zhao

felixendres, 2016-10-13:
Sorry, I don't know how to fix this. I'd guess the bug link and changes by @kbalisciano would give you a start.

owner01, 2016-10-17:
Also experiencing the same issue when trying to install on raspberry running ROS kinetic

owner01, 2016-10-17:
Fixed it by doing it in every file for the correct header file

felixendres, 2016-10-19:
If possible, please share your changes in a pull request

owner01, 2016-10-31:
Pff unfortunately i didn't write down which files I changed and am not using it anymore but I will ook for a backup. I know you have to add:
#ifndef Q_MOC_RUN
library giving error
#endif
To every file getting an error.
I know most of the time it had something to do with either QT or ros.h

blankered, 2016-10-31:
Can you push you changes to git? We can analyze it there.

owner01, 2016-11-09:
Unfortunately I did not backup those files

lhzlhz, 2017-11-07:
http://blog.csdn.net/h321654/article/details/54582341
The steps:
sudo gedit /usr/include/boost/type_traits/detail/has_binary_operator.hpp
Modify "namespace BOOST_JOIN(BOOST_TT_TRAIT_NAME,_impl) {...}" to
"
#ifndef Q_MOC_RUN
namespace BOOST_JOIN(BOOST_TT_TRAIT_NAME,_impl) {
#endif
....
#ifndef Q_MOC_RUN
}
#endif
"