felixendres/rgbdslam_v2 #22: Compiling requirements

melvinisken, 2015-11-09:
This is not really an issue, but I didn't know how to communicate this differently. I just wanted to mention that I was not able to compile the package with 1 GB of RAM. The compiler is always killed with an out of memory error. Increasing RAM to 2 GB seems to fix it. I don't know if it is worth to mention this in the ReadMe somewhere. (Maybe it's also just dependend of my individual configuration)

kbalisciano, 2015-11-09:
I have 2GB of RAM and 4GB of swap space but I still run out if I run it with 8 jobs. When you run make you can run make -j1 to make on one core. It takes hours but it will work

felixendres, 2015-11-09:
My gut feeling is that PCL is the culprit of the high memory requirements. Maybe this could be improved using precompiled headers?

kbalisciano, 2015-11-09:
If it crashes on me again I will let you know what it was building at that point. 

Felix did you get my pull request with the QT MOC fix for boost?

felixendres, 2015-11-09:
I just did. I'll comment it in a minute.

kbalisciano, 2015-12-01:
For me it's the CXX objects when you're working with the QT MOC that sucks up the memory.