felixendres/rgbdslam_v2 #23: Problem of Compiling rgdslam_v2 hydro 12.04

jayden118, 2015-11-24:
Hi all,

I'm trying to compile the code in ubuntu 12.04 with hydro ROS. However, the compilation is failed and shown the errors as follows:

make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
src/SiftGPU/CuTexImage.cpp:33:18: fatal error: cuda.h: No such file or directory
compilation terminated.
Linking CXX executable /home/jayden/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam
make[3]: *** [build/CuTexImage.o] Error 1
make[2]: *** [rgbdslam_v2-hydro/external/siftgpu_prefix/src/siftgpu_proj-stamp/siftgpu_proj-build] Error 2
make[1]: *** [rgbdslam_v2-hydro/external/CMakeFiles/siftgpu_proj.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lsiftgpu
collect2: ld returned 1 exit status
make[2]: *** [/home/jayden/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam] Error 1
make[1]: *** [rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed

Could you please give me some hint on how to solve this problem.

Thank you in advanced.

Regards,
Jayden

kbalisciano, 2015-11-24:
Turn off sift in the CMake file and it should compile.

jayden118, 2015-11-24:
Hi Kbalisciano,

Thank you for your prompt response. 

As I am new to ROS, could you please guide me more details on how to turn off the sift in cmake file?

Thank you.

kbalisciano, 2015-11-25:
sudo gedit ~/rgbdslam_catkin_ws/src/rgbdslam_v2/CMakeLists.txt

-set(USE_SIFT_GPU        1) 
+set(USE_SIFT_GPU        0) 

is typical I believe. It really depends on where your package is

felixendres, 2015-12-15:
I looked into this today. The automatic cuda-check of siftgpu doesn't work, so it is trying to use cuda anyway. I hardcoded cuda to be disabled in the makefile of siftgpu for hydro and indigo, so compilation should work again .

closed 2015-12-15 by felixendres