felixendres/rgbdslam_v2 #24: rosdep problem

kaygudo, 2015-12-27:
rosdep install rgbdslam_v2 yeilds:

ERROR: Rosdep cannot find all required resources to answer your query
Missing resource rgbdslam_v2
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks

felixendres, 2015-12-27:
Should be `rosdep install rgbdslam` without "_v2"

kaygudo, 2015-12-28:
i m sorry the problem was with my catkin workspace setup. The package
worked from me amazingly :) . How does the package does localization? it
would b very kind of you.

with regards

felixendres, 2016-01-05:
Please read the file README.md for usage information for the software and Chapter 3 of my [thesis](https://www.freidok.uni-freiburg.de/data/10239) if you'd like a description of the used algorithms.

closed 2016-01-05 by felixendres