felixendres/rgbdslam_v2 #26: No devices found (kinect works w freenect-glview)

jacksonkr, 2016-02-01:
rgbdslam worked fine on precise. I upgraded to trusty, installed ros indigo, and have everything working fine except that the camera isn't detected. You can see the output below.

```
$ roslaunch rgbdslam openni+rgbdslam.launch 
... logging to /home/jackson/.ros/log/41df8eaa-c913-11e5-be97-b886879b31ee/roslaunch-jackson-HP-Pavilion-Notebook-22023.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jackson-HP-Pavilion-Notebook:52593/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_hw/max_range: 4.0
 * /camera/disparity_registered_hw/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rgbdslam/config/backend_solver: pcg
 * /rgbdslam/config/cloud_creation_skip_step: 2
 * /rgbdslam/config/cloud_display_type: POINTS
 * /rgbdslam/config/detector_grid_resolution: 3
 * /rgbdslam/config/feature_detector_type: ORB
 * /rgbdslam/config/feature_extractor_type: ORB
 * /rgbdslam/config/max_keypoints: 600
 * /rgbdslam/config/max_matches: 300
 * /rgbdslam/config/min_sampled_candidates: 4
 * /rgbdslam/config/neighbor_candidates: 4
 * /rgbdslam/config/optimizer_skip_step: 1
 * /rgbdslam/config/pose_relative_to: largest_loop
 * /rgbdslam/config/predecessor_candidates: 4
 * /rgbdslam/config/ransac_iterations: 100
 * /rgbdslam/config/topic_image_depth: /camera/depth_reg...
 * /rgbdslam/config/topic_image_mono: /camera/rgb/image...
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    debayer (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_hw (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    rectify_color (nodelet/nodelet)
    rectify_ir (nodelet/nodelet)
    rectify_mono (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)
    rgbdslam (rgbdslam/rgbdslam)

auto-starting new master
process[master]: started with pid [22036]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 41df8eaa-c913-11e5-be97-b886879b31ee
process[rosout-1]: started with pid [22049]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [22066]
process[camera/driver-3]: started with pid [22067]
process[camera/debayer-4]: started with pid [22068]
process[camera/rectify_mono-5]: started with pid [22069]
process[camera/rectify_color-6]: started with pid [22079]
process[camera/rectify_ir-7]: started with pid [22091]
process[camera/depth_rectify_depth-8]: started with pid [22103]
[ INFO] [1454352026.660961647]: Initializing nodelet with 4 worker threads.
process[camera/depth_metric_rect-9]: started with pid [22117]
process[camera/depth_metric-10]: started with pid [22121]
process[camera/depth_points-11]: started with pid [22131]
process[camera/register_depth_rgb-12]: started with pid [22149]
process[camera/points_xyzrgb_sw_registered-13]: started with pid [22156]
process[camera/depth_registered_sw_metric_rect-14]: started with pid [22165]
process[camera/depth_registered_rectify_depth-15]: started with pid [22179]
process[camera/points_xyzrgb_hw_registered-16]: started with pid [22200]
process[camera/depth_registered_hw_metric_rect-17]: started with pid [22208]
process[camera/depth_registered_metric-18]: started with pid [22217]
process[camera/disparity_depth-19]: started with pid [22233]
process[camera/disparity_registered_sw-20]: started with pid [22236]
[ INFO] [1454352027.005966191]: No devices connected.... waiting for devices to be connected
process[camera/disparity_registered_hw-21]: started with pid [22247]
process[camera_base_link-22]: started with pid [22253]
process[camera_base_link1-23]: started with pid [22261]
process[camera_base_link2-24]: started with pid [22274]
process[camera_base_link3-25]: started with pid [22286]
process[rgbdslam-26]: started with pid [22301]
Initializing Node...
[ INFO] [1454352027.363848742]: Connected to roscore
[ INFO] [1454352027.776813671]: Using ORB keypoint detector.
[ INFO] [1454352027.776982264]: Using gridded keypoint detector with 3x3 cells, keeping 900 keypoints in total.
[ INFO] [1454352027.777047903]: Using adjusted keypoint detector with 5 maximum iterations, keeping the number of keypoints between 67 and 100
[ INFO] [1454352027.840839128]: Listening to /camera/rgb/image_color and /camera/depth_registered/sw_registered/image_rect_raw
[ INFO] [1454352030.007229077]: No devices connected.... waiting for devices to be connected
[ INFO] [1454352033.008288294]: No devices connected.... waiting for devices to be connected
[ INFO] [1454352036.008989125]: No devices connected.... waiting for devices to be connected

```

The kinect is listed in lsusb

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05c8:0379 Cheng Uei Precision Industry Co., Ltd (Foxlink) 
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 005: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 018: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 001 Device 016: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 001 Device 017: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 001 Device 015: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

and the kinect works with the freenect library. What should can I do to get the kinect working with rgbdslam w indigo?

kbalisciano, 2016-02-01:
Just looking at it briefly you should try to launch your camera node and slam separately. You'd be better off using freenect instead of openni. Also use the regular gl-view test to test out the Kinect connection prior to trying it in ros. I ended up making my own launch file

jacksonkr, 2016-02-01:
@kbalisciano regview works as well as the rest of the freenect demos. I'm not sure how to launch the camera node and slam separately. Do you know how I can accomplish that?

kbalisciano, 2016-02-01:
Launch your freenect.launch file from your freenect package and separately launch your rgbdslam launch file which should be subscribed to the same topics as the freenect is publishing. You don't want to try to launch the openni launch file when using freenect

jacksonkr, 2016-02-01:
@kbalisciano Worked like a charm, thanks for your wisdom.

```
$ roslaunch freenect_launch freenect.launch
$ roslaunch rgbdslam rgbdslam.launch
```

closed 2016-02-01 by felixendres

voyage19941119, 2017-05-19:
mine is worse ,  lsusb cannot find kinect ,but the light on kinect2 HUB is white(which means it is connected to the computer) . Do you have any idea ? I m confused

jacksonkr, 2017-05-19:
@voyage19941119 
I hate to break it to you but that's more of a "noob" question. I sympathize though, because I was there once. It might be that you have a libusb version that doesn't work with your kinect. You'll have to trial & error quite a bit until you find the issue. Good luck and don't forget to use google a LOT until you get over this hurdle. That's what I did.

voyage19941119, 2017-05-21:
sorry for the noob question～ First time using kinect ，thanks for helping though

voyage19941119, 2017-05-22:
@jacksonkr 
thanks again for reply.Now rgbd can receive pictures from my kinect . But ,just two frams ,then it stucked

[ WARN] [1495445549.453348471]: "openni_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1495445549.453373741]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)

I really have no idea about this after googling a lot ~

Have you changed the input topic value in rgbdslam.launch or openni+rgbdslam.launch?

jacksonkr, 2017-05-22:
@voyage19941119 I'll reach out to you separately as this is getting away from the original topic.