felixendres/rgbdslam_v2 #27: Following issues on El Captian (osx 10.11.3)

jacksonkr, 2016-02-08:
Original Post: http://answers.ros.org/question/225798/osx-rgbdslamv2-install-via-indigo/?comment=226057#post-id-226057

From there I've commented out libdevl in package.xml and now receiving the following error:

```
$ rosdep install rgbdslam
Error: No available formula with the name "ros/indigo/common_msgs" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/ros_comm" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/libg2o" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/message_runtime" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/std_msgs" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/geometry" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/perception_pcl" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/pcl_conversions" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/cmake_modules" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/vision_opencv" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/octomap" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/image_common" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "ros/indigo/message_generation" 
Please tap it and then try again: brew tap ros/indigo
Error: No available formula with the name "freeglut" 
Error: No available formula with the name "ros/indigo/catkin" 
Please tap it and then try again: brew tap ros/indigo
executing command [brew install ros/indigo/common_msgs]
==> Tapping ros/indigo
Cloning into '/usr/local/Library/Taps/ros/homebrew-indigo'...
Username for 'https://github.com': rollins.jackson@gmail.com
Password for 'https://rollins.jackson@gmail.com@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/ros/homebrew-indigo/' not found
Error: Failure while executing: git clone https://github.com/ros/homebrew-indigo /usr/local/Library/Taps/ros/homebrew-indigo --depth=1
ERROR: the following rosdeps failed to install
  homebrew: command [brew install ros/indigo/common_msgs] failed
```

I tried tapping ros/indigo:

```
$ brew tap ros/indigo
==> Tapping ros/indigo
Cloning into '/usr/local/Library/Taps/ros/homebrew-indigo'...
Username for 'https://github.com': rollins.jackson@gmail.com
Password for 'https://rollins.jackson@gmail.com@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/ros/homebrew-indigo/' not found
Error: Failure while executing: git clone https://github.com/ros/homebrew-indigo /usr/local/Library/Taps/ros/homebrew-indigo --depth=1
```

But you see that it's trying to get "ros/homebrew-indigo" - I researched for a while trying to find out how to get even just one package: "lib2go". Success yet to come. Any ideas?

felixendres, 2016-02-09:
No good ones. 

g2o support for MacOS is experimental, maybe there is no ROS package for g2o? You could build and install g2o from source and remove the catkin-dependency in package.xml

jacksonkr, 2016-02-09:
I downloaded g2o from `https://github.com/RainerKuemmerle/g2o` and was able to make it with `cmake .. -Wno-dev`. I'm still running into the long list of dependencies. ROS Indigo is installed and I can run rviz on osx el capitan. Getting rgbdslam running on el capitan is seeming like a tall order. I may have to wait until the dependencies work better with osx package managers (brew/macports).

felixendres, 2016-02-09:
I have no clue about ROS on OSX, but If not even ros_comm, std_msgs, and common_msgs work out, I suspect that something is wrong with your ROS setup. Did you set the environment variables right, e.g., ROS_PACKAGE_PATH?

jacksonkr, 2016-02-09:
Here's what I have:

```
$ printenv | grep ros
ROS_ROOT=/Users/Jackson/ros_catkin_ws/install_isolated/share/ros
ROS_PACKAGE_PATH=/Users/Jackson/rgbdslam_catkin_ws/src:/Users/Jackson/ros_catkin_ws/install_isolated/share:/Users/Jackson/ros_catkin_ws/install_isolated/stacks
OLDPWD=/Users/Jackson/ros_catkin_ws
CPATH=/Users/Jackson/rgbdslam_catkin_ws/devel/include:/Users/Jackson/ros_catkin_ws/install_isolated/include
PATH=/Users/Jackson/rgbdslam_catkin_ws/devel/bin:/Users/Jackson/ros_catkin_ws/install_isolated/bin:/usr/local/bin:/usr/local/sbin:/Users/Jackson/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin
DYLD_LIBRARY_PATH=/Users/Jackson/rgbdslam_catkin_ws/devel/lib:/Users/Jackson/ros_catkin_ws/install_isolated/lib
PYTHONPATH=/Users/Jackson/rgbdslam_catkin_ws/devel/lib/python2.7/site-packages:/Users/Jackson/ros_catkin_ws/install_isolated/lib/python2.7/site-packages:/usr/lib/python2.7
PKG_CONFIG_PATH=/Users/Jackson/rgbdslam_catkin_ws/devel/lib/pkgconfig:/Users/Jackson/ros_catkin_ws/install_isolated/lib/pkgconfig
CMAKE_PREFIX_PATH=/Users/Jackson/rgbdslam_catkin_ws/devel:/Users/Jackson/ros_catkin_ws/install_isolated
ROS_ETC_DIR=/Users/Jackson/ros_catkin_ws/install_isolated/etc/ros
```

felixendres, 2016-02-10:
Seems alright, as far as I can tell.

jacksonkr, 2016-02-10:
Agreed. I'm going to close this as unresolvable due to me being too newbie to solve the issue or El Capitan is not yet suited for rgbdslam. I suspect the problem is somewhere between the two.

closed 2016-02-10 by jacksonkr