felixendres/rgbdslam_v2 #29: RGBDSLAM Ros Indigo installation error - catkin_make error

ram651991, 2016-04-06:
With regards to this thread [https://github.com/felixendres/rgbdslam_v2/issues/13](url) , I get the following error while installing rgbdslam in ROS Indigo. Can anyone kindly help me out with issue ...

Running command: "make cmake_check_build_system" in "/home/ram/rgbdslam_catkin_ws/build"

####
####
Running command: "make -j4 -l4" in "/home/ram/rgbdslam_catkin_ws/build"

####
[ 0%] [ 0%] [ 0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[ 0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[ 8%] [ 13%] Built target rgbdslam_generate_messages_lisp
Built target rgbdslam_generate_messages_cpp
[ 31%] [ 34%] Built target rgbdslam_generate_messages_py
Built target siftgpu_proj
[ 34%] Built target rgbdslam_gencpp
[ 34%] Built target rgbdslam_generate_messages
[ 36%] [ 37%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o
[ 39%] [ 40%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/glviewer.cpp.o
Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/parameter_server.cpp.o
Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/node.cpp.o
virtual memory exhausted: Cannot allocate memory
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-4.8/README.Bugs for instructions.
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/glviewer.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-4.8/README.Bugs for instructions.
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o] Error 4
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

zazics, 2016-04-06:
When building on my Odroid XU4, I had same error and resolve it with "-j1" instead of "-j'4".
Probably too much memory used.

ram651991, 2016-04-06:
It threw error again, without "virtual memory exhausted: Cannot allocate memory" line and while building at 8% itself... :-/

> [  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
[  6%] Built target rgbdslam_generate_messages_cpp
[  6%] Built target rgbdslam_gencpp
[  8%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o] Error 4
make[1]: *** [rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2

zazics, 2016-04-06:
did you clean your build directory and reboot ?

ram651991, 2016-04-06:
I didn't do that the first time but after I did that, my laptop started to hang at 53% of build. I tried that thrice, and it was the same result.

zazics, 2016-04-07:
it hangs or crashes ? Maybe it's still working, slowly because of huge memory and CPU usage. Try again and be patient. Sometimes, I had to wait hours for a build to complete.

ram651991, 2016-04-07:
:-D ... Alright let me try that too, but apparently I had the same laptop with which I had rgbdslam working perfect on Ubuntu 12.04 and Ros Hydro. Anyways I'll get back if i'm done or not :+1:

ram651991, 2016-04-19:
I couldn't succeed in this :-/..... is there any other way i could possibly make it work indigo? Or can i install ROS Hydro successfully through some other means in Ubuntu 14.04???

ram651991, 2016-04-19:
I finally made it work :+1: ..... so my actual trouble was with the memory as @zazics had mentioned. I lately understood that the laptop in which I was working didn't have a swap memory and the RAM was just 2gb. So I created swap area with my pendrive and finally solved the make error. Now rgbdslam works fine for me!!!!!!!!!!!!

closed 2016-04-19 by ram651991