felixendres/rgbdslam_v2 #31: RGBDSLAM v2 errors on catkin_make

KaiserUDM, 2016-05-01:
I'm basically having this same issue http://answers.opencv.org/question/34814/help-with-headercode-adaptation-of-rgbdslam/
I can't find a way to resolve it for the life of me. I have edit the files to include #include and I get the same errors every time. Any insight into this issue would be amazing

felixendres, 2016-05-02:
The main problem I see for porting rgbdslam_v2 to OpenCV3, is that the code in feature_adjuster.* uses OpenCV2 base classes that do not exist in OpenCV3. Everything else is of minor difficulty.

The "right" way to handle this would be to write a general threshold adjuster and gridded detector as the one in the mentioned file and propose it as a patch to OpenCV. I have no idea why they got rid of the original functionality (except that it was very bad for processing of image streams), but I think it is of general interest, as most people still set fixed thresholds.

KaiserUDM, 2016-05-03:
I just started getting these errors on Sunday. Took me a while to realize that the base class was missing.  Also to fix my issue I just changed all the cv::vectors to std::vector. That shouldn't break anything correct?

felixendres, 2016-05-03:
As far as I know, that works. By the way, there is not much functionality 
provided by these Adjuster base classes that is not overridden. So maybe they 
can be changed to directly inherit from FeatureDetector.

revilokeb, 2016-08-16:
@KaiserUDM I am also running on OpenCV3.1 (Ubuntu 14.04 / indigo). Consequentially I am also getting cv::Vector / AdjusterAdapter errors. Have you managed to write a general threshold adjuster? Do the above adjustments go through, i.e. did you get rgbdslam running with OpenCV3?

felixendres, 2016-08-17:
I quickly tried search and replace _cv::AdjusterAdapter -> DetectorAdjuster_ and _cv::vector -> std::vector_, which solved the initial compiler errors. However, I ran into new problems, which I currently haven't got the time to solve.

bruderjakob12, 2016-10-24:
Any update on the issue?

Bis123, 2016-11-01:
I have the same issue.
Right now I have installed openCv 2.4.8 .
I am getting the following errors.

/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:40: error: ‘cv::vector’ has not been declared
  void detect(const cv::Mat& image, cv::vector<cv::KeyPoint>& keypoints, std::vector<float>& descriptors, const cv::Mat& mask = cv::Mat()) const;
                                        ^
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:46: error: expected ‘,’ or ‘...’ before ‘<’ token
  void detect(const cv::Mat& image, cv::vector<cv::KeyPoint>& keypoints, std::vector<float>& descriptors, const cv::Mat& mask = cv::Mat()) const;
                                              ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp: In constructor ‘Node::Node(const cv::Mat&, const cv::Mat&, const cv::Mat&, const CameraInfoConstPtr&, myHeader, cv::Ptr<cv::Feature2D>, cv::Ptr<cv::Feature2D>)’:
/home/SLAM/ws/src/rgbdslam/src/node.cpp:144:43: error: ‘cvtColor’ was not declared in this scope
     cvtColor(visual, gray_img, CV_RGB2GRAY);
                                           ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp:155:65: error: no matching function for call to ‘SiftGPUWrapper::detect(cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<float>&)’
     siftgpu->detect(gray_img, feature_locations_2d_, descriptors);
                                                                 ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp:155:65: note: candidate is:
In file included from /home/sfm/Bisna/SLAM/ws/src/rgbdslam/src/node.cpp:32:0:
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note: void SiftGPUWrapper::detect(const cv::Mat&, int) const
  void detect(const cv::Mat& image, cv::vector<cv::KeyPoint>& keypoints, std::vector<float>& descriptors, const cv::Mat& mask = cv::Mat()) const;
       ^
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note:   candidate expects 2 arguments, 3 provided
/home/SLAM/ws/src/rgbdslam/src/node.cpp:174:67: error: no matching function for call to ‘SiftGPUWrapper::detect(cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<float>&)’
       siftgpu->detect(gray_img, feature_locations_2d_, descriptors);
                                                                   ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp:174:67: note: candidate is:
In file included from /home/sfm/Bisna/SLAM/ws/src/rgbdslam/src/node.cpp:32:0:
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note: void SiftGPUWrapper::detect(const cv::Mat&, int) const
  void detect(const cv::Mat& image, cv::vector<cv::KeyPoint>& keypoints, std::vector<float>& descriptors, const cv::Mat& mask = cv::Mat()) const;
       ^
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note:   candidate expects 2 arguments, 3 provided
/home/SLAM/ws/src/rgbdslam/src/node.cpp: In constructor ‘Node::Node(cv::Mat, cv::Ptr<cv::Feature2D>, cv::Ptr<cv::Feature2D>, pcl::PointCloud<pcl::PointXYZRGB>::Ptr, cv::Mat)’:
/homeSLAM/ws/src/rgbdslam/src/node.cpp:278:71: error: ‘cvtColor’ was not declared in this scope
   if(visual.type() == CV_8UC3){ cvtColor(visual, gray_img, CV_RGB2GRAY); } 
                                                                       ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp:287:65: error: no matching function for call to ‘SiftGPUWrapper::detect(cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<float>&)’
     siftgpu->detect(gray_img, feature_locations_2d_, descriptors);
                                                                 ^
/home/SLAM/ws/src/rgbdslam/src/node.cpp:287:65: note: candidate is:
In file included from /home/sfm/Bisna/SLAM/ws/src/rgbdslam/src/node.cpp:32:0:
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note: void SiftGPUWrapper::detect(const cv::Mat&, int) const
  void detect(const cv::Mat& image, cv::vector<cv::KeyPoint>& keypoints, std::vector<float>& descriptors, const cv::Mat& mask = cv::Mat()) const;
       ^
/home/SLAM/ws/src/rgbdslam/src/sift_gpu_wrapper.h:49:7: note:   candidate expects 2 arguments, 3 provided
In file included from /home/sfm/Bisna/SLAM/ws/src/rgbdslam/src/misc.cpp:41:0:
/home/SLAM/ws/src/rgbdslam/src/aorb.h:65:56: error: ‘vector’ has not been declared
     void operator()(InputArray image, InputArray mask, vector<KeyPoint>& keypoints) const;
                                                        ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:65:62: error: expected ‘,’ or ‘...’ before ‘<’ token
     void operator()(InputArray image, InputArray mask, vector<KeyPoint>& keypoints) const;
                                                              ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:68:57: error: ‘vector’ has not been declared
     void operator()( InputArray image, InputArray mask, vector<KeyPoint>& keypoints,
                                                         ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:68:63: error: expected ‘,’ or ‘...’ before ‘<’ token
     void operator()( InputArray image, InputArray mask, vector<KeyPoint>& keypoints,
                                                               ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:68:10: error: ‘void cv::AORB::operator()(cv::InputArray, cv::InputArray, int) const’ cannot be overloaded
     void operator()( InputArray image, InputArray mask, vector<KeyPoint>& keypoints,
          ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:65:10: error: with ‘void cv::AORB::operator()(cv::InputArray, cv::InputArray, int) const’
     void operator()(InputArray image, InputArray mask, vector<KeyPoint>& keypoints) const;
          ^
[... cut ...]
     void computeImpl( const Mat& image, vector<KeyPoint>& keypoints, Mat& descriptors ) const;
                                               ^
/home/SLAM/ws/src/rgbdslam/src/aorb.h:76:40: error: ‘vector’ has not been declared
     void detectImpl( const Mat& image, vector<KeyPoint>& keypoints, const Mat& mask=Mat() ) const;
                                        ^
/home/
SLAM/ws/src/rgbdslam/src/aorb.h:76:46: error: expected ‘,’ or ‘...’ before ‘<’ token
     void detectImpl( const Mat& image, vector<KeyPoint>& keypoints, const Mat& mask=Mat() ) const;

felixendres, 2016-11-07:
Actually this looks like you are not building against OpenCV 2.4.8, but 3.0.

blankered, 2016-11-08:
@felixendres I have been working through the compiler errors with RGBD_v2 on my system which is Ubuntu 16 with ROS Kinetic, QT5, and openCV3. I have gotten pretty far but I can't find an alternative to the AdjusterAdapter member because OpenCV3 I guess did away with it. I am unable to find an alternative to this one hiccup.

felixendres, 2016-11-09:
IIRC this is a major hiccup. Anyway, I would give it another go. Would you share your changes, so I can try to work from there towards an OpenCV3 reimplementation of the Gridded Adjuster?

blankered, 2016-11-11:
@felixendres I have merely implemented your changes to fix the compiler errors. @fradelg fork looks as though he has completed much more in terms of making changes to update to QT5 and OpenCV3. I tried compiling his fork but had many compiler errors. If you want, I can try to debug the compiler errors from his version and then post his fork.

felixendres, 2016-11-20:
I have tried to migrate to qt5, opencv3. See the branch "kinetic" (it probably isn't kinetic-specific, but I have only kinetic installed). It compiles fine, but crashes on startup!
The problem is the one of issue #8:
- g2o requires code to be built with c++11
- pcl will cause a SIGSEV on startup if code is built with the c++11 flag
- The central class (and others) of rgbdslam has pcl and g2o includes

Therefore rgbdslam cannot work without changing either g2o or pcl. 

In #8 is a description how to make it work by recompiling pcl with C++11. I didn't do that yet, so I don't know whether my changes actually work. It would be great if someone could try to take this further.

felixendres, 2016-11-29:
I made a c++03 fork of g2o (see my g2o fork, branch "c++03") and updated the rgbdslam_v2 kinetic branch, so you can export the environment variable $G2O_DIR to where you installed the g2o from that branch to (e.g., using make install, possibly with $CMAKE_INSTALL_PREFIX env. var. set). Then build rgbdslam_v2 as usual.

While it compiles and "runs" there is some OpenCV error which I couldn't solve for now. Maybe next week.

psembere, 2016-12-01:
I have also problems to build RGBD-SLAM from branch Kinetic on clean Ubuntu 16.04 with ROS Kinetic.

When I try to build custom g2o c++03 fork from https://github.com/felixendres/g2o/tree/c++03 you mentioned in this commit c26e9b7b0c4595664fe48c0807a44a2ec9d69f66  I have errors printed below:

>Scanning dependencies of target circle_fit
[ 71%] Building CXX object g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o
In file included from /usr/include/c++/5/random:35:0,
                 from /home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:38,
                 from /home/ubuntu/catkin_ws/src/g2o/g2o/examples/data_fitting/circle_fit.cpp:32:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /home/ubuntu/catkin_ws/src/g2o/g2o/examples/data_fitting/circle_fit.cpp:32:0:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:44:71: error: ‘std::mt19937’ has not been declared
   double G2O_STUFF_API sampleUniform(double min=0, double max=1, std::mt19937* generator=0);
                                                                       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:45:39: error: ‘mt19937’ is not a member of ‘std’
   double G2O_STUFF_API sampleGaussian(std::mt19937* generator = 0);
                                       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:45:53: error: ‘generator’ was not declared in this scope
   double G2O_STUFF_API sampleGaussian(std::mt19937* generator = 0);
                                                     ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:83:10: error: ‘mt19937’ in namespace ‘std’ does not name a type
     std::mt19937* _generator;
          ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In constructor ‘g2o::GaussianSampler<SampleType, CovarianceType>::GaussianSampler(bool)’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:51:7: error: class ‘g2o::GaussianSampler<SampleType, CovarianceType>’ does not have any field named ‘_generator’
       _generator(0)
       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:54:9: error: ‘_generator’ was not declared in this scope
         _generator = new std::mt19937;
         ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:54:31: error: ‘mt19937’ in namespace ‘std’ does not name a type
         _generator = new std::mt19937;
                               ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In destructor ‘g2o::GaussianSampler<SampleType, CovarianceType>::~GaussianSampler()’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:58:14: error: ‘_generator’ was not declared in this scope
       delete _generator;
              ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In member function ‘SampleType g2o::GaussianSampler<SampleType, CovarianceType>::generateSample()’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:17: error: ‘_generator’ was not declared in this scope
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                 ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:56: error: ‘g2o::sampleGaussian’ cannot be used as a function
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                                                        ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:75: error: ‘g2o::sampleGaussian’ cannot be used as a function
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                                                                           ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In member function ‘bool g2o::GaussianSampler<SampleType, CovarianceType>::seed(int)’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:76:12: error: ‘_generator’ was not declared in this scope
       if (!_generator)
            ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:78:7: error: ‘_generator’ was not declared in this scope
       _generator->seed(s);
       ^
g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/build.make:62: recipe for target 'g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o' failed
make[2]: *** [g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o] Error 1
CMakeFiles/Makefile2:1795: recipe for target 'g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/all' failed
make[1]: *** [g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

felixendres, 2016-12-01:
Ah, sorry. You have to disable building of some parts. Set G2O_BUILD_EXAMPLES to false (e.g., with `cmake .. -DG2O_BUILD_EXAMPLES=OFF`).

psembere, 2016-12-02:
@felixendres Thanks. Meanwhile I tested RGBD-SLAM with PCL 1.8 compiled with c++11 and it is also crashing. 

When I have only custom G2O installed  (without kinetic version) I had a problem with:

> make[2]: *** No rule to make target '/opt/ros/kinetic/lib/libg2o_csparse_extension.so', needed by '/home/ubuntu/catkin_ws/devel/lib/rgbdslam/rgbdslam'.  Stop.

when I installed the ros-kinetic-libg2o it is fixed, so it is probably some dependency problem.

Using your advice I finally managed to build and run RGBD-SLAM, thanks again, however when I plugged my Asus Xtion I have got an error:

> [ WARN] [1480637640.285420397]: First RGBD-Data Received
[ INFO] [1480637640.285912291]: Encoding: rgb8
OpenCV Error: The function/feature is not implemented () in detectAndCompute, file /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/features2d/src/feature2d.cpp, line 144
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/features2d/src/feature2d.cpp:144: error: (-213)  in function detectAndCompute
[ INFO] [1480637640.404338179]: Stopping color stream.
================================================================================REQUIRED process [rgbdslam-16] has died!
process has died [pid 21903, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/ubuntu/.ros/log/382469a2-b824-11e6-9c05-000c293bf35e/rgbdslam-16.log].
log file: /home/ubuntu/.ros/log/382469a2-b824-11e6-9c05-000c293bf35e/rgbdslam-16*.log
Initiating shutdown!
================================================================================

So it's probably some OpenCV 3.1 compatibility issue.

felixendres, 2016-12-02:
Yes, probably. The error message I get is slightly different: The function/feature is not implemented () in detectAndCompute". I adapted the code in features.cpp, feature_adapter.cpp and aorb.cpp according to the [OpenCV migration guide ](http://docs.opencv.org/3.0-rc1/db/dfa/tutorial_transition_guide.html#tutorial_transition_algorithm) in commit 84a03c9cd990e257af6e1ad3cc6cc7d7686e16c4

That being done I am in the dark, so any help resolving this issue is appreciated.

felixendres, 2016-12-03:
I just identified the error source: The signature of the virtual methods of FeatureDetector (typedef'd now to Feature2D) has changed. Thus my subclass overrides do not override anymore and the base class's Feature2D::detect is called. Straightforward changing of the signature results in (hopefully) minor errors. I'll fix it in the coming week.

psembere, 2016-12-04:
Thank you very much.

felixendres, 2016-12-05:
Please try the newest commit.

psembere, 2016-12-06:
The applications starts but it hangs on adding elements to graph. 

> [ WARN] [1481035158.077103527]: First RGBD-Data Received
[ INFO] [1481035158.077766899]: Encoding: rgb8
[ INFO] [1481035158.078772072]: Depth and RGB image off more than 1/30sec: 967466592 (nsec)
[ INFO] [1481035158.099100728]: Feature 2d size: 497, 3D: 497
[ INFO] [1481035158.099166256]: Keypoints: 497
[ INFO] [1481035158.099189431]: Feature Count of Node:	497
[ WARN] [1481035158.109608344]: "openni_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1481035158.109678599]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)
[ INFO] [1481035158.113014826]: Ground Truth Transform for First Node: Translation 0 0 0
[ INFO] [1481035158.113071434]: Ground Truth Transform for First Node: Rotation 0 0 0 1
[ INFO] [1481035158.113100352]: Adding initial node with id 0 and seq 0, v_id: 0
[ INFO] [1481035158.114353400]: Encoding: rgb8
[ INFO] [1481035158.120921969]: Setting Renderable
[ INFO] [1481035158.127096959]: No change to renderer skip (1).
[ INFO] [1481035158.127234234]: 1 Keyframes: 0, 
[ INFO] [1481035158.139972766]: Feature 2d size: 554, 3D: 554
[ INFO] [1481035158.140014799]: Keypoints: 554
[ INFO] [1481035158.140024821]: Feature Count of Node:	554
[ WARN] [1481035158.150289642]: "openni_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1481035158.150345525]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)
[ INFO] [1481035158.152750248]: Node ID's to compare with candidate for node 1. Sequential: 0
[ INFO] [1481035158.152815434]: Nodes to compare: 0, 
[ INFO] [1481035158.152846658]: Running node comparisons in parallel in 3 (of 4) available threads
[ INFO] [1481035158.152995469]: Encoding: rgb8
[ INFO] [1481035158.153479734]: You specified FLANN with ORB, this is SLOW! Using BRUTEFORCE instead.
[ INFO] [1481035158.165622365]: My bruteforce Search runtime: 0.0118533 s
[ INFO] [1481035158.167752378]: Feature 2d size: 447, 3D: 447
[ INFO] [1481035158.167775733]: Keypoints: 447
[ INFO] [1481035158.167882307]: Feature Count of Node:	447
[ INFO] [1481035158.171533513]: 48 good iterations (from 100), inlier pct 70, inlier cnt: 211, error (MHD): 0.74
[ INFO] [1481035158.171599035]: Nodes 1<->0: 48 good iterations (from 100), inlier pct 70, inlier cnt: 211, error (MHD): 0.74
[ INFO] [1481035158.171623848]: RANSAC found a valid transformation with 211 inliers matches with average ratio 0.111691
[ INFO] [1481035158.171639123]: Returning Valid Edge
[ INFO] [1481035158.171705919]: Result of comparison 0: 0(
[ INFO] [1481035158.171734595]: Rotation: 0.15, Distance:  0.001m
[ INFO] [1481035158.171750918]: Rotation: 0.15, Distance:  0.001m
[ INFO] [1481035158.171879996]: Added Edge between 0 and 1. Inliers: 211
[ INFO] [1481035158.171914181]: Adding node with id 1 and seq id 1 to the graph
[ INFO] [1481035158.171931590]: create cloud 1 1 0
[ INFO] [1481035158.176972395]: No change to renderer skip (1).
[ INFO] [1481035158.177228714]: Added Node, new graphsize: 2 nodes
[ INFO] [1481035158.177360146]: Added, Camera Pose Graph Size: 2N/1E, Duration: 0.024633, Inliers:  211
[ WARN] [1481035158.178745450]: Loop Closures: 0, Sequential Edges: 1
[ WARN] [1481035158.178775437]: Starting Optimization
[ WARN] [1481035158.178845479]: Optimization with 2 cams, 140236587801344 nodes and 140236587806496 edges in the graph
[ INFO] [1481035158.178935651]: optimizeGraphImpl runtime: 0.00137832 s
[ WARN] [1481035158.179491658]: "openni_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1481035158.179506709]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)
[ INFO] [1481035158.196694159]: Node ID's to compare with candidate for node 2. Sequential: 0, 1
[ INFO] [1481035158.196809989]: Nodes to compare: 1, 0, 
[ INFO] [1481035158.196834076]: Running node comparisons in parallel in 3 (of 4) available threads
[ INFO] [1481035158.196871708]: You specified FLANN with ORB, this is SLOW! Using BRUTEFORCE instead.
[ INFO] [1481035158.197289937]: You specified FLANN with ORB, this is SLOW! Using BRUTEFORCE instead.
[ INFO] [1481035158.198716549]: Encoding: rgb8
[ INFO] [1481035158.202810510]: My bruteforce Search runtime: 0.00545525 s
[ INFO] [1481035158.203001343]: My bruteforce Search runtime: 0.00592115 s
[ INFO] [1481035158.204442257]: Last Resort: Trying identity as hypothesis
[ INFO] [1481035158.204505940]: 0 good iterations (from 100), inlier pct 0, inlier cnt: 0, error (MHD): 1000000.00
[ INFO] [1481035158.204525190]: Nodes 2<->0: 0 good iterations (from 100), inlier pct 0, inlier cnt: 0, error (MHD): 1000000.00
[ INFO] [1481035158.204542882]: RANSAC found no valid trafo, but had initially 300 feature matches with average ratio 0.218952
[ INFO] [1481035158.204624571]: Last Resort: Trying identity as hypothesis
[ INFO] [1481035158.204695115]: 0 good iterations (from 100), inlier pct 0, inlier cnt: 0, error (MHD): 1000000.00
[ INFO] [1481035158.204778694]: Nodes 2<->1: 0 good iterations (from 100), inlier pct 0, inlier cnt: 0, error (MHD): 1000000.00
[ INFO] [1481035158.204853582]: RANSAC found no valid trafo, but had initially 300 feature matches with average ratio 0.215010
[... cut ...]
[ WARN] [1481035158.205111357]: No valid (sequential) transformation between 1 and 2: Using constant position assumption.
[ INFO] [1481035158.219703550]: Feature 2d size: 415, 3D: 415
[ INFO] [1481035158.219772092]: Keypoints: 415
[ INFO] [1481035158.219790228]: Feature Count of Node:	415
[ WARN] [1481035158.230010428]: "openni_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1481035158.230108799]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)

psembere, 2016-12-06:
Ok, I think it is working now, I had a mess with g2o, because export G2O_DIR=~/g2ofork/install does not work, I tried to set include_directories it worked, but now it doesn't :D

felixendres, 2016-12-06:
Hm, I have not installed ros-kinetic-libg2o and it builds fine. Have you tried deleting your g2o build folder and rebuilding g2o from scratch?
Another idea is that, maybe when you deinstalled ros-kinetic-libg2o, you also deinstalled some dependencies of g2o, like libsuitesparse-dev?

psembere, 2016-12-07:
Thank you very much, after I removed the  ros-kinetic-libg2o and rebuild g2o afterwards it finally worked :D  No linker errors. But the G2O_DIR flag does not work.

felixendres, 2016-12-15:
Could you guys please try the new install script (see README.md "Installation from scratch" in kinetic branch). It works fine on my machine, including the usage of the G2O_DIR environment variable.

closed 2016-12-15 by felixendres

arunumd, 2018-12-07:
Dear @felixendres I installed PCL Version 1.8 and configured rgbdslam to look for the new PCL version in _/usr/local/lib_ but I still get this strange error. My rgbdslam starts momentarily and then dies all together, by throwing the error message below : 
```
================================================================================REQUIRED process [rgbdslam-35] has died!
process has died [pid 1216, exit code -11, cmd /home/arun/Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam /camera/rgb/image_color:=/camera/rgb/image_raw /camera/depth_registered/sw_registered/image_rect_raw:=/camera/depth/image_raw __name:=rgbdslam __log:=/home/arun/.ros/log/0767c1dc-fa68-11e8-86d6-5cea1d834771/rgbdslam-35.log].
log file: /home/arun/.ros/log/0767c1dc-fa68-11e8-86d6-5cea1d834771/rgbdslam-35*.log
Initiating shutdown!
================================================================================

```
Can you please guide me what could be the issue here ?

HaiQuan0303, 2020-09-05:
i tried to install but i still can't

commit 3d5e8520 (2016-12-15): Install script and its documentation.

Fixes #31