felixendres/rgbdslam_v2 #32: rosdep install rgbdslam error in indigo

brightlight2084, 2016-06-04:
My ros version is indigo. And my rosdep version is newest(0.11.5-1).
1. I do the command "rosdep update" and it finished with no error.
2. I do the command "rosdep install rgbdslam", and then a error is displayed.
the details is below:

w@w-Aspire-VN7-791G:~/catkin_ws$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
updated cache in /home/wangke/.ros/rosdep/sources.cache
w@w-Aspire-VN7-791G:~/catkin_ws$ rosdep install rgbdslam

ERROR: Rosdep cannot find all required resources to answer your query
Missing resource rgbdslam
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks

brightlight2084, 2016-06-04:
use the commands below, and success.

#Prepare Workspace
source /opt/ros/indigo/setup.bash
mkdir -p ~/rgbdslam_catkin_ws/src
cd ~/rgbdslam_catkin_ws/src
catkin_init_workspace
cd ~/rgbdslam_catkin_ws/
catkin_make
source devel/setup.bash

#Get RGBDSLAM
cd ~/rgbdslam_catkin_ws/src
wget -q http://github.com/felixendres/rgbdslam_v2/archive/indigo.zip
unzip -q indigo.zip
cd ~/rgbdslam_catkin_ws/

#Install
rosdep update
rosdep install rgbdslam
catkin_make

closed 2016-06-04 by brightlight2084

felixendres, 2016-06-06:
> ERROR: Rosdep cannot find all required resources to answer your query
Missing resource rgbdslam
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks

This means it can't find rgbdslam itself. By the printed paths it is apparent that ROS doesn't know about your workspace yet.

Therefore you just needed to source your workspace setup script, i.e., in your workspace directory issue `source devel/setup.bash`

Since you did that in your second try, it was succesful.