felixendres/rgbdslam_v2 #35: no cloud points showing

QinZiwen, 2016-07-02:
I am using ubuntu 14.04, ros inidgo and kinect2. I had installed libfreenect2 and iai_kinect2, then I tested them which run nice.But when I "roslaunch rgbdslam.launch ", the terminal show some messages following and I do not see cloud points.
+++++++++++++++++++++++++++++
....
auto-starting new master
process[master]: started with pid [7915]
ROS_MASTER_URI=http://turtlebot-HP-ZBook-14-G2.local:11311

setting /run_id to a39bcbd6-4032-11e6-a1d1-6480999b1afe
process[rosout-1]: started with pid [7928]
started core service [/rosout]
process[rgbdslam-2]: started with pid [7945]
Initializing Node...
[ INFO] [1467449692.428582021]: Connected to roscore
[ INFO] [1467449692.581739930]: Using ORB keypoint detector.
[ INFO] [1467449692.581822911]: Using gridded keypoint detector with 3x3 cells, keeping 900 keypoints in total.
[ INFO] [1467449692.581859952]: Using adjusted keypoint detector with 5 maximum iterations, keeping the number of keypoints between 67 and 100
[ INFO] [1467449692.602606555]: Listening to /kinect2/sd/image_color_rect and /kinect2/sd/image_depth_rect
....
+++++++++++++++++++++++++++++
what should I do? 
Thanks

felixendres, 2016-07-05:
Is the ros master on a different machine? If the images are transmitted over a network there are often problems with dropped messages, so that no RGB-D pairs with matching timestamps are received. This is particularly problematic over wireless lan. In that case you have two options: Throttle the driver node or switch to qvga. There are other possibilities if that doesn't help.

QinZiwen, 2016-07-06:
Thank you @felixendres . RGBDSLAM can work, when I run ros master and RGBDSLAM in identical machine.But I find that windows of procedure  seem show  single frame result fusion. When I move camera few distance, point cloud does not augment. Terminal show warnning as following:
+++++++++++++++++++++++++++++++
[ WARN] [1467815104.738537648]: Time jump (time delta: 47.23)
[ WARN] [1467815104.738597656]: Did not add as Node
[ WARN] [1467815104.821219674]: "kinect2_rgb_optical_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1467815104.821299079]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as transformation (This message is throttled to 1 per 5 seconds)
+++++++++++++++++++++++++++++++

I am using ubuntu 14.04, ros inidgo and kinect2. I had installed libfreenect2 and iai_kinect2. The sequence from kinect2 been translated by kinect2_bridge not openni2. Because openni2 does not work in my machine.

felixendres, 2016-07-13:
> [ WARN] [1467815104.738537648]: Time jump (time delta: 47.23)

This is weird. Are there any irregularities with your timestamps?

> [ WARN] [1467815104.821219674]: "kinect2_rgb_optical_frame" passed to lookupTransform argument source_frame does not exist.
> [ WARN] [1467815104.821299079]: Using Standard kinect /openni_camera -> /openni_rgb_optical_frame as tran sformation (This message is throttled to 1 per 5 seconds)

This shouldn't be a problem. Probably the sensor frames are called differently by the driver. You can get rid of this message by setting the paramter "base_frame_name" to the frame_id stated in your images (see `rostopic echo /kinect2/sd/image_color_rect/header`).

QinZiwen, 2016-07-14:
Because I use libfreenect2 and iai_kinect2, so I change the rgbdslam.launch as follows:
   
    
    <param name="camera_info_topic"                    value="/kinect2/qhd/camera_info"/>
	<param name="config/topic_image_mono"              value="/kinect2/qhd/image_mono_rect"/> 
    <param name="config/topic_image_depth"             value="/kinect2/qhd/image_depth_rect"/>
    <param name="config/topic_points"                  value="/kinect2/qhd/points"/> 

when I type "roslaunch rgbdslam rgbdslam.launch", the error show:
Initializing Node...
[ INFO] [1468462053.117081285]: Connected to roscore
[ INFO] [1468462053.269145001]: Using ORB keypoint detector.
[ INFO] [1468462053.269204174]: Using gridded keypoint detector with 3x3 cells, keeping 900 keypoints in total.
[ INFO] [1468462053.269236003]: Using adjusted keypoint detector with 5 maximum iterations, keeping the number of keypoints between 67 and 100
[ INFO] [1468462053.290941353]: Listening to /kinect2/qhd/image_mono_rect, /kinect2/qhd/image_depth_rect and /kinect2/qhd/points
[ WARN] [1468462054.374146160]: First RGBD-Data Received
[ INFO] [1468462054.407852939]: Construction of Node with ORB Features
[ INFO] [1468462054.422055136]: Feature 2d size: 794, 3D: 600
[ INFO] [1468462054.422185834]: Feature 2d size: 600, 3D: 600
rgbdslam: /home/turtlebot/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/src/node.cpp:315: Node::Node(cv::Mat, cv::Ptr<cv::FeatureDetector>, cv::Ptr<cv::DescriptorExtractor>, pcl::PointCloud<pcl::PointXYZRGB>::Ptr, cv::Mat): Assertion `feature_locations_2d_.size() == feature_locations_3d_.size()' failed.
================================================================================REQUIRED process [rgbdslam-1] has died!
process has died [pid 32330, exit code -6, cmd /home/turtlebot/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/turtlebot/.ros/log/1cf4dd76-4964-11e6-8423-6480999b1afe/rgbdslam-1.log].
log file: /home/turtlebot/.ros/log/1cf4dd76-4964-11e6-8423-6480999b1afe/rgbdslam-1*.log
initiating shutdown!
================================================================================R
[rgbdslam-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

felixendres, 2016-07-14:
Ok, this is definitely a bug, thanks for reporting. I'll have to fix that, but you can work around this by setting the parameter "config/topic_points" to "" (i.e., an empty string).

QinZiwen, 2016-07-14:
ok, Thank you very much @felixendres