felixendres/rgbdslam_v2 #36: http://wiki.ros.org/rgbdslam_v2 has nothing

QinZiwen, 2016-07-06:
When I look "http://wiki.ros.org/rgbdslam_v2", nothing show me.

felixendres, 2016-07-15:
Feel free to write it. Probably you can copy most (all?) of the readme there.

130s, 2016-07-15:
I've asked for the package to be indexed https://github.com/ros/rosdistro/pull/12038 for Indigo, so at minimum the content of package.xml will be automatically copied soon.

QinZiwen, 2016-07-16:
Thank you @felixendres @130s

130s, 2016-07-28:
@dirk-thomas It's been almost a month since https://github.com/ros/rosdistro/pull/12038 was merged but I don't see the document is generated yet on http://wiki.ros.org/rgbdslam_v2. 
Where can we see the doc job results for debugging?

dirk-thomas, 2016-07-28:
The job has generated the documented successfully: http://build.ros.org/job/Idoc__rgbdslam_v2__ubuntu_trusty_amd64/

The repository contains only a single package named `rgbdslam` (not `rgbdslam_v2`): https://github.com/felixendres/rgbdslam_v2/blob/76d8b8911f0a6ed0f7a051b220187fe5089d2a5e/package.xml#L3

So the generated information can be found on http://wiki.ros.org/rgbdslam

130s, 2016-07-28:
Ah, I see :D
Should [rosdistro entry](https://github.com/ros/rosdistro/pull/12038/files#diff-4bbb8e35ff5dc5e7f2d7b4d00374533fR8624) need to be changed from rgbdslam_v2 to rgbdslam as well?

dirk-thomas, 2016-07-28:
It could - but it doesn't change what is being generated - only the job names.

130s, 2016-07-28:
Thanks for clarification @dirk-thomas. Sounds like the rosdistro entry name can be different from the package name, so I think we can leave it as it is.

@felixendres @QinZiwen I've updated http://wiki.ros.org/rgbdslam_v2 to navigate to http://wiki.ros.org/rgbdslam. Also I added a link to readme on this repo from rgbdslam page, hoping that's good enough. I'll leave the rest up to you all.

felixendres, 2016-08-11:
Thanks a lot @130s and @dirk-thomas

closed 2016-08-11 by felixendres