felixendres/rgbdslam_v2 #37: Running RGB-D SLAM on the Benchmark dataset

jpeger, 2016-09-20:
hi, I try to run rgbdslam on TUM benchmark dataset

benchmark directory includes .bag and groundtruth.txt, 
such as, rgbd_dataset_freiburg1_xyz.bag and rgbd_dataset_freiburg1_xyz-groundtruth.txt

command : $ rosrun rgbdslam benchmark_evaluation.sh SURF path_to_dataset_directory

=================================================================
here are error messages, 
...
Evaluating SIFTGPU
Will evaluate RGBD-SLAM on the following bagfiles:
ls: cannot access rgbd_dataset_freiburg2_large_no_loop.bag: No such file or directory

'' is not a directory
Usage: /home/myname/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/rgbd_benchmark/summarize_evaluation.sh <directory-where-results-are>
e.g., /home/myname/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/rgbd_benchmark/summarize_evaluation.sh some/path/to/SURF/
Evaluating SIFTGPU
Will evaluate RGBD-SLAM on the following bagfiles:
ls: cannot access rgbd_dataset_freiburg2_large_no_loop.bag: No such file or directory

'' is not a directory
Usage: /home/myname/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/rgbd_benchmark/summarize_evaluation.sh <directory-where-results-are>
e.g., /home/myname/rgbdslam_catkin_ws/src/rgbdslam_v2-indigo/rgbd_benchmark/summarize_evaluation.sh some/path/to/SURF/
==================================================================

ref: http://wiki.ros.org/rgbdslam_electric/evaluation

thanks a lot

felixendres, 2016-09-21:
`roslaunch rgbdslam test_settings.launch bagfile_name:="/your/path/to/the/TUM/bagfile.bag"` should run rgbdslam with default parameters for the benchmark dataset. Use the menu to save the point cloud manually when the processing finished.

jpeger, 2016-09-22:
by command
    roslaunch rgbdslam test_settings.launch bagfile_name:="/your/path/to/the/TUM/bagfile.bag"
such as,
    roslaunch rgbdslam test_settings.launch bagfile_name:=rgbd_dataset_freiburg1_xyz.bag

I have a problem,
==============================================================================
[ INFO] Time:[1474509594.373093478] Thread:[7f12b3fff700]: Loading Bagfile rgbd_dataset_freiburg1_xyz.bag
[FATAL] Time:[1474509594.373295443] Thread:[7f12b3fff700]: Opening Bagfile rgbd_dataset_freiburg1_xyz.bag failed: Error opening file: rgbd_dataset_freiburg1_xyz.bag Quitting!
==============================================================================
REQUIRED process [rgbdslam-2] has died!
process has finished cleanly
log file: /home/ray/.ros/log/4125e164-8068-11e6-bbd4-448a5b9fc942/rgbdslam-2*.log
Initiating shutdown!
==============================================================================

I check .bag data is usable by 
    roscore + rosbag play rgbd_dataset_freiburg1_xyz.bag. It worrks.

felixendres, 2016-09-22:
You need the absolute path of your bagfile, /home/ray/.../rgbd_dataset_freiburg1_xyz.bag

jpeger, 2016-09-23:
It works!
Thank a lot, again!

closed 2016-09-26 by felixendres

voyage19941119, 2017-05-22:
I run testsettings.launch by    roslaunch rgbdslam test_settings.launch bagfile_name:="mypath/TUM/bagfile.bag" ,   and it stucked after processing 10 frames ? Does the robot in the bag file moves too fast for the rgbdslam? or is my cpu too slow?

Dhagash4, 2017-09-27:
`roslaunch rgbdslam test_settings.launch bagfile_name:="/home/catkin_ws/src/rgbdslam_v2/test/rgbd_dataset_freiburg1_desk.bag.1"`
I am using roscore + above code still I am getting the same error.

Below is the error please help me to resolve it.....

> [FATAL] Time:[1506502655.647540635] Thread:[7f8aebfff700]: Opening Bagfile rgbd_dataset_freiburg1_desk.bag.1 failed: Error opening file: rgbd_dataset_freiburg1_desk.bag.1 Quitting!
================================================================================REQUIRED process [rgbdslam-1] has died!
process has finished cleanly
log file: /home/dhagash/.ros/log/d162e17a-a360-11e7-b814-4cbb58d55d91/rgbdslam-1*.log
Initiating shutdown!

@felixendres

Dhagash4, 2017-09-27:
Got it thanks