felixendres/rgbdslam_v2 #38: rgbdslam_v2 fail at the start up. Error in `/home/username/catkin_ws/devel/lib/rgbdslam/rgbdslam': realloc(): invalid pointer: ...

Negvet, 2016-09-22:
I have Ubuntu 14.04, ROS Indigo.

I downloaded rgbdslam_v2-indigo and placed it into catkin_ws/src. catkin_ws initialized and builded before. I installed all dependencies with rosdep and then did catkin_make. It finished with
> [ 98%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/features.cpp.o
> [100%] Building CXX object rgbdslam_v2-indigo/CMakeFiles/rgbdslam.dir/src/sift_gpu_wrapper.cpp.o
> Linking CXX executable /home/evgeny/catkin_ws/devel/lib/rgbdslam/rgbdslam
> [100%] Built target rgbdslam

I have new Xtion,  so I installed openni2-launch instead of openni-launch (openni-launch did not work, it pop up USB interface error). I tested it with the rviz - it works.

I am starting openni node separately with 
> roslaunch openni_launch openni.launch

Now the question! When I run 
> rosrun rgbdslam rgbdslam

I got the following error:
> *** Error in `/home/evgeny/catkin_ws/devel/lib/rgbdslam/rgbdslam': realloc(): invalid pointer: 0x00007f1ba0d51840 ***

The pointer number is different all the times.
I tried to run catkin_make again. I tried "catkin_make clean" and then build again. Always the same error. 

Help please!

felixendres, 2016-09-22:
Hi, thanks for reporting. I don't know what that error could be caused by. Could you please add the whole output, not only the one line with the error.

If possible, please try `rosrun --prefix 'gdb -ex run --args' rgbdslam rgbdslam`, then after the crash type `backtrace` and attach the output here.

Negvet, 2016-09-22:
> roslaunch rgbdslam openni+rgbdslam.launch

return the following:

> ... logging to /home/evgeny/.ros/log/a72a6aec-80c8-11e6-9a7a-d85de252fcd5/roslaunch-asus-3180.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://asus:38754/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera/camera_nodelet_manager/num_worker_threads: 4
>  * /camera/depth_rectify_depth/interpolation: 0
>  * /camera/depth_registered_rectify_depth/interpolation: 0
>  * /camera/disparity_depth/max_range: 4.0
>  * /camera/disparity_depth/min_range: 0.5
>  * /camera/disparity_registered_hw/max_range: 4.0
>  * /camera/disparity_registered_hw/min_range: 0.5
>  * /camera/disparity_registered_sw/max_range: 4.0
>  * /camera/disparity_registered_sw/min_range: 0.5
>  * /camera/driver/depth_camera_info_url: 
>  * /camera/driver/depth_frame_id: camera_depth_opti...
>  * /camera/driver/depth_registration: False
>  * /camera/driver/device_id: #1
>  * /camera/driver/rgb_camera_info_url: 
>  * /camera/driver/rgb_frame_id: camera_rgb_optica...
>  * /rgbdslam/config/backend_solver: pcg
>  * /rgbdslam/config/cloud_creation_skip_step: 2
>  * /rgbdslam/config/cloud_display_type: POINTS
>  * /rgbdslam/config/detector_grid_resolution: 3
>  * /rgbdslam/config/feature_detector_type: ORB
>  * /rgbdslam/config/feature_extractor_type: ORB
>  * /rgbdslam/config/max_keypoints: 600
>  * /rgbdslam/config/max_matches: 300
>  * /rgbdslam/config/min_sampled_candidates: 4
>  * /rgbdslam/config/neighbor_candidates: 4
>  * /rgbdslam/config/optimizer_skip_step: 1
>  * /rgbdslam/config/pose_relative_to: largest_loop
>  * /rgbdslam/config/predecessor_candidates: 4
>  * /rgbdslam/config/ransac_iterations: 100
>  * /rgbdslam/config/topic_image_depth: /camera/depth_reg...
>  * /rgbdslam/config/topic_image_mono: /camera/rgb/image...
>  * /rosdistro: indigo
>  * /rosversion: 1.11.20
> 
> NODES
>   /camera/
>     camera_nodelet_manager (nodelet/nodelet)
>     debayer (nodelet/nodelet)
>     depth_metric (nodelet/nodelet)
>     depth_metric_rect (nodelet/nodelet)
>     depth_points (nodelet/nodelet)
>     depth_rectify_depth (nodelet/nodelet)
>     depth_registered_hw_metric_rect (nodelet/nodelet)
>     depth_registered_metric (nodelet/nodelet)
>     depth_registered_rectify_depth (nodelet/nodelet)
>     depth_registered_sw_metric_rect (nodelet/nodelet)
>     disparity_depth (nodelet/nodelet)
>     disparity_registered_hw (nodelet/nodelet)
>     disparity_registered_sw (nodelet/nodelet)
>     driver (nodelet/nodelet)
>     points_xyzrgb_hw_registered (nodelet/nodelet)
>     points_xyzrgb_sw_registered (nodelet/nodelet)
>     rectify_color (nodelet/nodelet)
>     rectify_ir (nodelet/nodelet)
>     rectify_mono (nodelet/nodelet)
[... 162 lines cut ...]
> (gdb) backtrace
> #0  0x00007ffff0ddbc37 in __GI_raise (sig=sig@entry=6)
>     at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> #1  0x00007ffff0ddf028 in __GI_abort () at abort.c:89
> #2  0x00007ffff0e182a4 in __libc_message (do_abort=do_abort@entry=1, 
>     fmt=fmt@entry=0x7ffff0f266b0 "*** Error in `%s': %s: 0x%s ***\n")
>     at ../sysdeps/posix/libc_fatal.c:175
> #3  0x00007ffff0e22ff7 in malloc_printerr (action=<optimized out>, 
>     str=0x7ffff0f22945 "realloc(): invalid pointer", ptr=<optimized out>)
>     at malloc.c:4996
> #4  0x00007ffff0e280a2 in __GI___libc_realloc (
>     oldmem=0x7ffff24e0840 <QListData::shared_null>, bytes=32) at malloc.c:2986
> #5  0x00007fffe24ade3e in QListData::realloc(int) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #6  0x00007fffe24adefa in QListData::append(int) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #7  0x00007fffe259ad8c in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #8  0x00007fffe2596e29 in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #9  0x00007fffe2439eb3 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #10 0x00007ffff7dea10a in call_init (l=<optimized out>, argc=argc@entry=1, 
>     argv=argv@entry=0x7fffffffdb48, env=env@entry=0x7fffffffdb58)
>     at dl-init.c:78
> ---Type <return> to continue, or q <return> to quit---
>

Negvet, 2016-09-22:
> #11 0x00007ffff7dea1f3 in call_init (env=<optimized out>, 
>     argv=<optimized out>, argc=<optimized out>, l=<optimized out>)
>     at dl-init.c:36
> #12 _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffdb48, 
>     env=0x7fffffffdb58) at dl-init.c:126
> #13 0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
> #14 0x0000000000000001 in ?? ()
> #15 0x00007fffffffdf5c in ?? ()
> #16 0x0000000000000000 in ?? ()

felixendres, 2016-09-22:
rgbdslam [requires qt4](https://github.com/felixendres/rgbdslam_v2/blob/indigo/CMakeLists.txt#L98), you have qt5. I don't know why it would build correctly and link against qt5.
Maybe issue #17 has more information?

Negvet, 2016-09-22:
I deleted qt5.

> evgeny@asus:~$ qmake --version
> QMake version 2.01a
> Using Qt version 4.8.6 in /usr/lib/x86_64-linux-gnu

but I have the same errors:

> evgeny@asus:~/rgbdslam_catkin_ws$ rosrun --prefix 'gdb -ex run --args' rgbdslam rgbdslam
> GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
> Copyright (C) 2014 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "x86_64-linux-gnu".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from /home/evgeny/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam...done.
> Starting program: /home/evgeny/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam 
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> *** Error in `/home/evgeny/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam': realloc(): invalid pointer: 0x00007ffff24e0840 ***
> 
> Program received signal SIGABRT, Aborted.
> 0x00007ffff0ddbc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> 56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
> (gdb) backtrace
> #0  0x00007ffff0ddbc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> #1  0x00007ffff0ddf028 in __GI_abort () at abort.c:89
> #2  0x00007ffff0e182a4 in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff0f266b0 "*** Error in `%s': %s: 0x%s ***\n")
>     at ../sysdeps/posix/libc_fatal.c:175
> #3  0x00007ffff0e22ff7 in malloc_printerr (action=<optimized out>, str=0x7ffff0f22945 "realloc(): invalid pointer", ptr=<optimized out>)
>     at malloc.c:4996
> #4  0x00007ffff0e280a2 in __GI___libc_realloc (oldmem=0x7ffff24e0840 <QListData::shared_null>, bytes=32) at malloc.c:2986
> #5  0x00007fffe24ade3e in QListData::realloc(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #6  0x00007fffe24adefa in QListData::append(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #7  0x00007fffe259ad8c in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #8  0x00007fffe2596e29 in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #9  0x00007fffe2439eb3 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #10 0x00007ffff7dea10a in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffdc08, env=env@entry=0x7fffffffdc18)
>     at dl-init.c:78
> #11 0x00007ffff7dea1f3 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:36
> #12 _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffdc08, env=0x7fffffffdc18) at dl-init.c:126
> #13 0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
> #14 0x0000000000000001 in ?? ()
> #15 0x00007fffffffdffd in ?? ()
> #16 0x0000000000000000 in ?? ()
> (gdb) 
> #0  0x00007ffff0ddbc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
> #1  0x00007ffff0ddf028 in __GI_abort () at abort.c:89
> #2  0x00007ffff0e182a4 in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff0f266b0 "*** Error in `%s': %s: 0x%s ***\n")
>     at ../sysdeps/posix/libc_fatal.c:175
> #3  0x00007ffff0e22ff7 in malloc_printerr (action=<optimized out>, str=0x7ffff0f22945 "realloc(): invalid pointer", ptr=<optimized out>)
>     at malloc.c:4996
> #4  0x00007ffff0e280a2 in __GI___libc_realloc (oldmem=0x7ffff24e0840 <QListData::shared_null>, bytes=32) at malloc.c:2986
> #5  0x00007fffe24ade3e in QListData::realloc(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #6  0x00007fffe24adefa in QListData::append(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #7  0x00007fffe259ad8c in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #8  0x00007fffe2596e29 in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) ()
>    from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #9  0x00007fffe2439eb3 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
> #10 0x00007ffff7dea10a in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffdc08, env=env@entry=0x7fffffffdc18)
>     at dl-init.c:78
> #11 0x00007ffff7dea1f3 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:36
> #12 _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffdc08, env=0x7fffffffdc18) at dl-init.c:126
> #13 0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
> #14 0x0000000000000001 in ?? ()
> #15 0x00007fffffffdffd in ?? ()
> #16 0x0000000000000000 in ?? ()

felixendres, 2016-09-22:
The error still occurs in _/usr/lib/x86_64-linux-gnu/libQt5Core.so.5_ (in levels 5 to 9 of the backtrace). Have you cleaned an rebuilt? Try `rm -rf ~/rgbdslam_catkin_ws/build/` before catkin_make, to make sure cmake is not caching anything qt5 related.

Negvet, 2016-09-27:
I completely reinstall ROS but it did't help. Looks like there still some qt5 related packages that participated in the making process (which I could't find). I removed all files from /usr/lib/x86_64-linux-gnu/ with "Qt5" in the name, but it did not help.

Actually my problem is that I'm not that good at Linux, therefore the only things that solved my issue is reinstalling the Ubuntu. rgbdslam fit default ubuntu 14.04 pretty well.

Thanks!

felixendres, 2016-09-27:
Sorry to hear you had to take the brute force approach (i.e. reinstall).

closed 2016-09-27 by felixendres

Negvet, 2016-09-28:
no problem. 
just one more comment: yes, on default Ubuntu I did catkin_make and everything works fine. But after installing all dependencies for OpenCV (qt5-default among them) and OpenCV inself, I have the same error while launching rgbdslam:

> Error in `/home/negvet/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam': realloc(): invalid pointer: 0x00007f9fc000a840

looks like I just need to configure rgbdslam to not to use everything related to qt5 while starting up, but I do not know how..

felixendres, 2016-09-30:
Please check whether you built against the right version. `ldd /path/to/your/executable/rgbdslam` should turn up something like

    (...)
    libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4
    (...)

If there is a 5 instead of 4 you can try the following: Check out the exact qt 4 version with `/usr/bin/qmake-qt4 -v`. For me this prints `Using Qt version 4.8.6`.

Then state the version exactly in the CMakeLists.txt of rgbdslam by changing the line

    find_package(Qt4 REQUIRED)

to reflect the exact version, e.g., 

    find_package(Qt4 4.8.6 EXACT REQUIRED)

Negvet, 2016-10-03:
`ldd /path/to/your/executable/rgbdslam` return 

```
...
libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f1510a11000)
libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f15103c4000)
libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f150fb9b000)
libQt5Test.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Test.so.5 (0x00007f150f971000)
libQt5OpenGL.so.5 => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f150f70d000)
...
libQtOpenGL.so.4 => /usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4 (0x00007f1521812000)
libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f1520b5f000)
libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f1520679000)
...
```

So here are both 4 and 5.
`/usr/bin/qmake-qt4 -v` returns 
```
QMake version 2.01a
Using Qt version 4.8.6 in /usr/lib/x86_64-linux-gnu
```

So I updated CMakeLists.txt with 
`find_package(Qt4 4.8.6 EXACT REQUIRED)`

Then I run `catkin_make clean` , `rm -rf ~/rgbdslam_catkin_ws/build/`,  `catkin_make` but finally I have the same error:
`Error in /home/negvet/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam': realloc(): invalid pointer: 0x00007f51a375a840``

felixendres, 2016-10-05:
Hm, weird. Maybe another library is pulling in qt5? I'm out of ideas how to solve this, sorry.