felixendres/rgbdslam_v2 #39: Problem making catkin_make

patrilopez, 2016-11-17:
Hi, i tried to install rgbdslam, but i can't make catkin_make.I have these errors :
Linking CXX executable /home/patri/catkin_ws/devel/lib/rgbdslam/rgbdslam
/usr/bin/ld:  cannot find -lQt5::Core
/usr/bin/ld:  cannot find -lQt5::Gui
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld:  cannot findr -lQt5::Test
/usr/bin/ld:  cannot find -lQt5::Concurrent
/usr/bin/ld:  cannot find -lQt5::OpenGL
collect2: error: ld returned 1 exit status
make[2]: *** [/home/patri/catkin_ws/devel/lib/rgbdslam/rgbdslam] Error 1
make[1]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Help, please!

felixendres, 2016-11-20:
Hi Patrilopez. Could you elaborate on which versions of ROS, ubuntu, and pcl you have?
You could try the new "kinetic" branch, which solves the issue you have, but depending on your PCL version you will run into other problems.

psembere, 2016-12-01:
I have also problems to build RGBD-SLAM from branch Kinetic on clean Ubuntu 16.04 with ROS Kinetic.

When I try to build custom g2o c++03 fork from https://github.com/felixendres/g2o/tree/c++03 you mentioned in this commit c26e9b7b0c4595664fe48c0807a44a2ec9d69f66  I have errors printed below:

>Scanning dependencies of target circle_fit
[ 71%] Building CXX object g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o
In file included from /usr/include/c++/5/random:35:0,
                 from /home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:38,
                 from /home/ubuntu/catkin_ws/src/g2o/g2o/examples/data_fitting/circle_fit.cpp:32:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /home/ubuntu/catkin_ws/src/g2o/g2o/examples/data_fitting/circle_fit.cpp:32:0:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:44:71: error: ‘std::mt19937’ has not been declared
   double G2O_STUFF_API sampleUniform(double min=0, double max=1, std::mt19937* generator=0);
                                                                       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:45:39: error: ‘mt19937’ is not a member of ‘std’
   double G2O_STUFF_API sampleGaussian(std::mt19937* generator = 0);
                                       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:45:53: error: ‘generator’ was not declared in this scope
   double G2O_STUFF_API sampleGaussian(std::mt19937* generator = 0);
                                                     ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:83:10: error: ‘mt19937’ in namespace ‘std’ does not name a type
     std::mt19937* _generator;
          ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In constructor ‘g2o::GaussianSampler<SampleType, CovarianceType>::GaussianSampler(bool)’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:51:7: error: class ‘g2o::GaussianSampler<SampleType, CovarianceType>’ does not have any field named ‘_generator’
       _generator(0)
       ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:54:9: error: ‘_generator’ was not declared in this scope
         _generator = new std::mt19937;
         ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:54:31: error: ‘mt19937’ in namespace ‘std’ does not name a type
         _generator = new std::mt19937;
                               ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In destructor ‘g2o::GaussianSampler<SampleType, CovarianceType>::~GaussianSampler()’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:58:14: error: ‘_generator’ was not declared in this scope
       delete _generator;
              ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In member function ‘SampleType g2o::GaussianSampler<SampleType, CovarianceType>::generateSample()’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:17: error: ‘_generator’ was not declared in this scope
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                 ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:56: error: ‘g2o::sampleGaussian’ cannot be used as a function
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                                                        ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:70:75: error: ‘g2o::sampleGaussian’ cannot be used as a function
         s(i) = (_generator) ? sampleGaussian(_generator) : sampleGaussian();
                                                                           ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h: In member function ‘bool g2o::GaussianSampler<SampleType, CovarianceType>::seed(int)’:
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:76:12: error: ‘_generator’ was not declared in this scope
       if (!_generator)
            ^
/home/ubuntu/catkin_ws/src/g2o/g2o/stuff/sampler.h:78:7: error: ‘_generator’ was not declared in this scope
       _generator->seed(s);
       ^
g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/build.make:62: recipe for target 'g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o' failed
make[2]: *** [g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/circle_fit.cpp.o] Error 1
CMakeFiles/Makefile2:1795: recipe for target 'g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/all' failed
make[1]: *** [g2o/examples/data_fitting/CMakeFiles/circle_fit.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Is this version really necessary? Does RGBD-SLAM code has to be c++03 compatible? Can I compile RGBD-SLAM with standard Kinetic g2o version and using c++11 compiler? Maybe I should use different compiler version like 4.x ?

When I used the default Kinetic g2o version and added to CMakeList.txt add_compile_options(-std=c++11) program compiles, however when I run it I get a message with pointer to log file that doesn't exist, so it is hard to debug this problem. Is this a problem with PCL version? I'm using 1.7.2 and  tried also custom 1.7.1 but it is the same.

`================================================================================REQUIRED process [rgbdslam-2] has died!
process has died [pid 2772, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/ubuntu/.ros/log/4f42391e-b7e0-11e6-870d-000c293bf35e/rgbdslam-2.log].
log file: /home/ubuntu/.ros/log/4f42391e-b7e0-11e6-870d-000c293bf35e/rgbdslam-2*.log
Initiating shutdown!
================================================================================`

xiteen, 2017-04-10:
Hi
I have the same problem when I try to build this package.
The environment is Ubuntu 14.04, ROS Ingido.
The problem report is following:

Base path: /home/ben/catkin_ws
Source space: /home/ben/catkin_ws/src
Build space: /home/ben/catkin_ws/build
Devel space: /home/ben/catkin_ws/devel
Install space: /home/ben/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ben/catkin_ws/build"
####
####
#### Running command: "make -j2 -l2" in "/home/ben/catkin_ws/build"
####
[  0%] [  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[ 13%] Built target siftgpu_proj
[ 20%] Built target rgbdslam_generate_messages_cpp
[ 28%] Built target rgbdslam_generate_messages_py
[ 35%] [ 35%] Built target rgbdslam_generate_messages_lisp
Built target rgbdslam_gencpp
[ 35%] Built target rgbdslam_generate_messages
[ 36%] [ 38%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o
Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/qt_gui.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

felixendres, 2017-04-10:
Yes, sounds like an out-of-memory kill.