felixendres/rgbdslam_v2 #41: catkin_make error in moc_glviewer

mayurand, 2016-12-09:
Hi

I m trying to make the package on ROS kinetic and Ubuntu 16.04. The pcl version I have is 1.7.2 and Qt 4.8. I tried several ways to make the package reading the issues in the repository. I initially had problems compiling the g2ofork. The c++03 branch finally worked for me using 
cmake .. -DCMAKE_INSTALL_PREFIX=~/g2ofork/install -DG2O_BUILD_EXAMPLES=OFF 

When making the rgbdslam_v2 however, I am having the following error. Request you to provide inputs on the same.

Cheers!

Error:

[ 52%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_qtros.cpp.o
[ 54%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o
[ 55%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o
[ 57%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_qt_gui.cpp.o
[ 58%] Building CXX object rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_glviewer.cpp.o
In file included from /usr/include/c++/5/unordered_map:35:0,
                 from /opt/ros/kinetic/include/g2o/core/hyper_graph.h:38,
                 from /opt/ros/kinetic/include/g2o/core/optimizable_graph.h:38,
                 from /opt/ros/kinetic/include/g2o/core/sparse_optimizer.h:32,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/graph_manager.h:58,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/openni_listener.h:31,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/moc_openni_listener.cpp:9:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/include/c++/5/unordered_map:35:0,
                 from /opt/ros/kinetic/include/g2o/core/hyper_graph.h:38,
                 from /opt/ros/kinetic/include/g2o/core/optimizable_graph.h:38,
                 from /opt/ros/kinetic/include/g2o/core/sparse_optimizer.h:32,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/graph_manager.h:58,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/moc_graph_manager.cpp:9:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /opt/ros/kinetic/include/g2o/core/optimizable_graph.h:38:0,
                 from /opt/ros/kinetic/include/g2o/core/sparse_optimizer.h:32,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/graph_manager.h:58,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/openni_listener.h:31,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/moc_openni_listener.cpp:9:
/opt/ros/kinetic/include/g2o/core/hyper_graph.h:138:20: error: ‘unordered_map’ in namespace ‘std’ does not name a template type
       typedef std::unordered_map<int, Vertex*>     VertexIDMap;
                    ^
/opt/ros/kinetic/include/g2o/core/hyper_graph.h:225:13: error: ‘VertexIDMap’ does not name a type
       const VertexIDMap& vertices() const {return _vertices;}
             ^
/opt/ros/kinetic/include/g2o/core/hyper_graph.h:227:7: error: ‘VertexIDMap’ does not name a type
       VertexIDMap& vertices() {return _vertices;}
       ^
/opt/ros/kinetic/include/g2o/core/hyper_graph.h:274:7: error: ‘VertexIDMap’ does not name a type
       VertexIDMap _vertices;
       ^
In file included from /opt/ros/kinetic/include/g2o/core/sparse_block_matrix.h:38:0,
                 from /opt/ros/kinetic/include/g2o/core/sparse_optimizer.h:33,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/graph_manager.h:58,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/openni_listener.h:31,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/moc_openni_listener.cpp:9:
/opt/ros/kinetic/include/g2o/core/sparse_block_matrix_ccs.h:222:20: error: ‘unordered_map’ in namespace ‘std’ does not name a template type
       typedef std::unordered_map<int, MatrixType*> SparseColumn;
                    ^
/opt/ros/kinetic/include/g2o/core/sparse_block_matrix_ccs.h:241:25: error: ‘SparseColumn’ was not declared in this scope
       const std::vector<SparseColumn>& blockCols() const { return _blockCols;}
                         ^
/opt/ros/kinetic/include/g2o/core/sparse_block_matrix_ccs.h:241:37: error: template argument 1 is invalid
       const std::vector<SparseColumn>& blockCols() const { return _blockCols;}
                                     ^
/opt/ros/kinetic/include/g2o/core/sparse_block_matrix_ccs.h:241:37: error: template argument 2 is invalid
/opt/ros/kinetic/include/g2o/core/sparse_block_matrix_ccs.h:242:19: error: ‘SparseColumn’ was not declared in this scope
       std::vector<SparseColumn>& blockCols() { return _blockCols;}
                   ^
[... 128 lines cut ...]
                ^
In file included from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/../../../src/rgbdslam_v2/src/graph_manager.h:65:0,
                 from /home/atros/rgbdslam_v2/build/rgbdslam_v2/src/moc_graph_manager.cpp:9:
/opt/ros/kinetic/include/g2o/core/robust_kernel_impl.h:49:45: error: ‘RobustKernelPtr’ does not name a type
       explicit RobustKernelScaleDelta(const RobustKernelPtr& kernel, double delta = 1.);
                                             ^
/opt/ros/kinetic/include/g2o/core/robust_kernel_impl.h:53:13: error: ‘RobustKernelPtr’ does not name a type
       const RobustKernelPtr kernel() const { return _kernel;}
             ^
/opt/ros/kinetic/include/g2o/core/robust_kernel_impl.h:55:28: error: ‘RobustKernelPtr’ does not name a type
       void setKernel(const RobustKernelPtr& ptr);
                            ^
/opt/ros/kinetic/include/g2o/core/robust_kernel_impl.h:60:7: error: ‘RobustKernelPtr’ does not name a type
       RobustKernelPtr _kernel;
       ^
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:110: recipe for target 'rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o' failed
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:158: recipe for target 'rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o' failed
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o] Error 1
CMakeFiles/Makefile2:3158: recipe for target 'rgbdslam_v2/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

felixendres, 2016-12-09:
Hi Mayur,
thanks for posting what works. I'll include the line
> cmake .. -DCMAKE_INSTALL_PREFIX=~/g2ofork/install -DG2O_BUILD_EXAMPLES=OFF

in the README.

The below part of your error message shows that the compiler uses the headers of the ros-kinetic-libg2o package. Please try to use the *BEFORE* parameter in [CMakeLists.txt line 92](https://github.com/felixendres/rgbdslam_v2/blob/kinetic/CMakeLists.txt#L92), so that it reads
`include_directories(BEFORE SYSTEM ${G2O_INCLUDE_DIR} ${G2O_INCLUDE_DIRS})`.
If that doesn't work (I would be very interested to know whether it does), I guess you must uninstall ros-kinetic-libg2o.

> In file included from /usr/include/c++/5/unordered_map:35:0,
> from /opt/ros/kinetic/include/g2o/core/hyper_graph.h:38,
> from /opt/ros/kinetic/include/g2o/core/optimizable_graph.h:38,
> from [...]

mayurand, 2016-12-09:
Hi,

For the g2o make, I would like to inform that if the packages libqglviewer-dev package (which is based on qt5) and libqglviewer-dev-qt4 are present then they break the compilation. I removed them and the g++03 fork worked.

I tried the _BEFORE_ argument but it didnt work. I had compiled the g2ofork without the ros-kinetic-libg2o. I tried catkin_make with and without the ros-kinetic-libg2o but in both the cases the same result.

anuppari, 2016-12-09:
I'm having the same issue. Can you explain exactly what you did? Did you compile g2o separately? How? What is the g++03 fork?

felixendres, 2016-12-10:
@mayurand, interesting. If the BEFORE didn't work, I have no idea other than uninstalling.

@anuppari, the g2o c++03 fork removes features only available in c++11, the C++  2011 standard and therefore falls back to the 2003 standard c++03. Apparently this is necessary to link g2o and boost (which pcl uses) to the same binary, i.e., rgbdslam_v2.

The [Installation From Scratch](https://github.com/felixendres/rgbdslam_v2/blob/kinetic/README.md#installation-from-scratch) section of the README.md file in the kinetic branch lists how to install rgbdslam_v2 with the g2o c++03 fork. 

Let me know if something is not accurate.

anuppari, 2016-12-10:
Yeah I didn't realize the instructions for kinetic were different. I
noticed a few typos in them and have opened another issue with details.

-Anup Parikh

mayurand, 2016-12-12:
@felixendres Thanks for your support. I tried to force the std c++11 by setting compiler in Cmakelists.txt (line 35). It did build the package but the run file does not execute. That results in segementation fault.

felixendres, 2016-12-12:
@mayurand, that is exactly why the c++03 fork exists.

felixendres, 2016-12-15:
I have added an install script for the kinetic brachn, that should help you. Have a look at the section "Installation from scratch" in README.md.

cumtchenchang, 2017-05-14:
sorry to disturb you. when i use make, i can not solver the problem, please help me
- Build files have been written to: /home/cc/catkin_ws/src/rgbdslam_v2/build
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  7%] Built target rgbdslam_generate_messages_py
[ 12%] Built target rgbdslam_generate_messages_nodejs
[ 18%] Built target rgbdslam_generate_messages_lisp
[ 25%] Built target rgbdslam_generate_messages_eus
[ 31%] Built target rgbdslam_generate_messages_cpp
[ 31%] Built target rgbdslam_generate_messages
[ 31%] Built target rgbdslam_gencpp
Scanning dependencies of target rgbdslam
[ 32%] Building CXX object CMakeFiles/rgbdslam.dir/src/moc_qtros.cpp.o
[ 34%] Building CXX object CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o
In file included from /usr/include/c++/5/unordered_map:35:0,
                 from /usr/local/include/g2o/core/hyper_graph.h:38,
                 from /usr/local/include/g2o/core/optimizable_graph.h:38,
                 from /usr/local/include/g2o/core/sparse_optimizer.h:32,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/graph_manager.h:58,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/openni_listener.h:31,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/moc_openni_listener.cpp:9:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/local/include/g2o/core/optimizable_graph.h:38:0,
                 from /usr/local/include/g2o/core/sparse_optimizer.h:32,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/graph_manager.h:58,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/openni_listener.h:31,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/moc_openni_listener.cpp:9:
/usr/local/include/g2o/core/hyper_graph.h:138:20: error: ‘unordered_map’ in namespace ‘std’ does not name a template type
       typedef std::unordered_map<int, Vertex*>     VertexIDMap;
                    ^
/usr/local/include/g2o/core/hyper_graph.h:225:13: error: ‘VertexIDMap’ does not name a type
       const VertexIDMap& vertices() const {return _vertices;}
             ^
/usr/local/include/g2o/core/hyper_graph.h:227:7: error: ‘VertexIDMap’ does not name a type
       VertexIDMap& vertices() {return _vertices;}
       ^
/usr/local/include/g2o/core/hyper_graph.h:274:7: error: ‘VertexIDMap’ does not name a type
       VertexIDMap _vertices;
       ^
In file included from /usr/local/include/g2o/core/sparse_block_matrix.h:38:0,
                 from /usr/local/include/g2o/core/sparse_optimizer.h:33,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/graph_manager.h:58,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/openni_listener.h:31,
                 from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/moc_openni_listener.cpp:9:
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:222:20: error: ‘unordered_map’ in namespace ‘std’ does not name a template type
       typedef std::unordered_map<int, MatrixType*> SparseColumn;
                    ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:241:25: error: ‘SparseColumn’ was not declared in this scope
       const std::vector<SparseColumn>& blockCols() const { return _blockCols;}
                         ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:241:37: error: template argument 1 is invalid
       const std::vector<SparseColumn>& blockCols() const { return _blockCols;}
                                     ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:241:37: error: template argument 2 is invalid
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:242:19: error: ‘SparseColumn’ was not declared in this scope
       std::vector<SparseColumn>& blockCols() { return _blockCols;}
                   ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:242:31: error: template argument 1 is invalid
       std::vector<SparseColumn>& blockCols() { return _blockCols;}
                               ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:242:31: error: template argument 2 is invalid
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:273:19: error: ‘SparseColumn’ was not declared in this scope
       std::vector<SparseColumn> _blockCols;     ///< the matrices stored in CCS order
                   ^
/usr/local/include/g2o/core/sparse_block_matrix_ccs.h:273:31: error: template argument 1 is invalid
       std::vector<SparseColumn> _blockCols;     ///< the matrices stored in CCS order
[... 28 lines cut ...]
/usr/local/include/g2o/core/robust_kernel.h: At global scope:
/usr/local/include/g2o/core/robust_kernel.h:74:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<RobustKernel> RobustKernelPtr;
                ^
In file included from /home/cc/catkin_ws/src/rgbdslam_v2/build/src/../../src/graph_manager.h:65:0,
[... cut ...]
/usr/local/include/g2o/core/robust_kernel_impl.h:60:7: error: ‘RobustKernelPtr’ does not name a type
       RobustKernelPtr _kernel;
       ^
CMakeFiles/rgbdslam.dir/build.make:110: recipe for target 'CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o' failed
make[2]: *** [CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o] Error 1
CMakeFiles/Makefile2:2936: recipe for target 'CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2