felixendres/rgbdslam_v2 #42: Installation instruction errors in kinetic branch

anuppari, 2016-12-09:
There are a few typos in the installation instructions for kinetic. 

1. Need to clone the `c++03` branch of `g2o`
2. Put clone in the correct folder (i.e., g2ofork in the below example)
3. Actually clone `rgbslam` instead of g2o twice
4. Remove ros-kinetic-libg2o

This should circumvent the qt/Boost parsing issues as well as g2o needing c++11. This works on ROS Kinetic and Ubuntu 16.04. Here's the updated version, changes in bold

<pre>
#Prepare Workspace
source /opt/ros/kinetic/setup.bash
mkdir -p ~/rgbdslam_catkin_ws/src
cd ~/rgbdslam_catkin_ws/src
catkin_init_workspace
cd ~/rgbdslam_catkin_ws/
catkin_make
source devel/setup.bash

#Get and build g2o fork
mkdir -p ~/g2ofork
<b>git clone -b c++03 https://github.com/felixendres/g2o.git g2ofork</b>
mkdir -p ~/g2ofork/build
cd ~/g2ofork/build
cmake .. -DCMAKE_INSTALL_PREFIX=~/g2ofork/install -DG2O_BUILD_EXAMPLES=OFF
make -j2 install

#Get and build rgbdslam_v2
export G2O_DIR=~/g2ofork/install
cd ~/rgbdslam_catkin_ws/src
git clone -b kinetic <b>https://github.com/felixendres/rgbdslam_v2.git</b>
cd ~/rgbdslam_catkin_ws/
rosdep update
rosdep install rgbdslam
<b>sudo apt-get purge ros-kinetic-libg2o</b>
catkin_make 
</pre>

anuppari, 2016-12-12:
`rosdep install rgbdslam` will re-download and install `ros-kinetic-libg2o`. Hence the package has to be purged after installing all dependencies.

felixendres, 2016-12-12:
Thanks, I guess I should remove libg2o from package.xml then. And next time I'll better do it, when I can actually verify it works :-)

felixendres, 2016-12-15:
Ok, done. I have put the installation sequence in a script that includes your feedback and verified it works on my machine. Thanks for your input.

closed 2016-12-15 by felixendres