felixendres/rgbdslam_v2 #43: Parse error at "BOOST_JOIN"

akosmaroy, 2016-12-11:
when compiling rgbdslam_v2,  get the following error:

```
$ catkin_make
Base path: /home/akos/src/ros/src/catkin_ws
Source space: /home/akos/src/ros/src/catkin_ws/src
Build space: /home/akos/src/ros/src/catkin_ws/build
Devel space: /home/akos/src/ros/src/catkin_ws/devel
Install space: /home/akos/src/ros/src/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/akos/src/ros/src/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/akos/src/ros/src/catkin_ws/build"
####
[  1%] Built target pico_flexx_driver_gencfg
[  2%] Performing build step for 'siftgpu_proj'
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  2%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  2%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[  2%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
[  2%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  5%] Built target pico_flexx_nodelet
[  8%] Built target pico_flexx_driver
[ 14%] Built target rgbdslam_generate_messages_lisp
[ 22%] Built target rgbdslam_generate_messages_py
[ 28%] Built target rgbdslam_generate_messages_cpp
[ 28%] Built target rgbdslam_generate_messages
[ 28%] Built target rgbdslam_gencpp
[ 31%] Generating src/moc_glviewer.cxx
[ 32%] Generating src/moc_ros_service_ui.cxx
[ 34%] Generating src/moc_openni_listener.cxx
[ 35%] Generating src/moc_qt_gui.cxx
[ 37%] Generating src/moc_graph_manager.cxx
[ 31%] Generating src/moc_qtros.cxx
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:62: recipe for target 'rgbdslam_v2/src/moc_qtros.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_qtros.cxx] Error 1
make[2]: *** Waiting for unfinished jobs....
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:87: recipe for target 'rgbdslam_v2/src/moc_ros_service_ui.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_ros_service_ui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:82: recipe for target 'rgbdslam_v2/src/moc_glviewer.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_glviewer.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:72: recipe for target 'rgbdslam_v2/src/moc_qt_gui.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_qt_gui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:67: recipe for target 'rgbdslam_v2/src/moc_openni_listener.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_openni_listener.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2/CMakeFiles/rgbdslam.dir/build.make:77: recipe for target 'rgbdslam_v2/src/moc_graph_manager.cxx' failed
make[2]: *** [rgbdslam_v2/src/moc_graph_manager.cxx] Error 1
CMakeFiles/Makefile2:1355: recipe for target 'rgbdslam_v2/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 38%] No install step for 'siftgpu_proj'
[ 40%] Completed 'siftgpu_proj'
[ 47%] Built target siftgpu_proj
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```

this is on Ubuntu 16.04 64 bit

felixendres, 2016-12-11:
Are you using ROS kinetic and accordingly the kinetic branch of rgbdslam?

akosmaroy, 2016-12-11:
I seem to be using indigo:

```
$ echo $ROS_DISTRO
indigo

```

felixendres, 2016-12-11:
I didn't know that works on 16.04. Anyway I haven't encountered the problem. Googling it hints to a conflict between specific versions of qt and boost. So you'd have to go to qt5 or up/downgrade to another boost version. You may try the kinetic branch of rgbdslam_v2, it should work with indigo (I haven't tested that though). The installation process is different, so make sure to follow the instructions in the README in that branch.

ahundt, 2016-12-12:
fyi indigo doesn't support 16.04, only kinetic does

LuisSerna, 2019-02-26:
had the same error and was solved with this line in the CMakeList
include_directories(include ${catkin_INCLUDE_DIRS})