felixendres/rgbdslam_v2 #44: cmake error when install rgbdslam v2 using install.sh

jackj012, 2016-12-15:
I used to install rgbdslam_v2 on my kinetic laptop successfully. however, i deleted the rgbdslam workspace for some reason  and now tried to install it again. I first followed instructions in this link: https://github.com/felixendres/rgbdslam_v2/issues/42
didn't work. Then i tried the install.sh, still didn't work. Both failed in cmake process. Here are the errors:

-- Searching for g2o ...
-- Searching for g2o in /home/jack/Code/g2ofork/install
-- Found g2o headers in: /home/jack/Code/g2ofork/install/include
-- Found libg2o: /home/jack/Code/g2ofork/install/lib/libg2o_csparse_extension.so;/home/jack/Code/g2ofork/install/lib/libg2o_core.so;/home/jack/Code/g2ofork/install/lib/libg2o_stuff.so;/home/jack/Code/g2ofork/install/lib/libg2o_types_slam3d.so;G2O_SOLVER_CHOLMOD_LIB-NOTFOUND;/home/jack/Code/g2ofork/install/lib/libg2o_solver_pcg.so;/home/jack/Code/g2ofork/install/lib/libg2o_solver_csparse.so
G2o-libraries /home/jack/Code/g2ofork/install/lib/libg2o_csparse_extension.so/home/jack/Code/g2ofork/install/lib/libg2o_core.so/home/jack/Code/g2ofork/install/lib/libg2o_stuff.so/home/jack/Code/g2ofork/install/lib/libg2o_types_slam3d.soG2O_SOLVER_CHOLMOD_LIB-NOTFOUND/home/jack/Code/g2ofork/install/lib/libg2o_solver_pcg.so/home/jack/Code/g2ofork/install/lib/libg2o_solver_csparse.so
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
/home/jack/Code/rgbdslam_catkin_ws/src/rgbdslam/external/SiftGPU/
/home/jack/Code/rgbdslam_catkin_ws/build/rgbdslam/external/siftgpu_prefix
Using OpenCV from 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
G2O_SOLVER_CHOLMOD_LIB
    linked by target "rgbdslam" in directory /home/jack/Code/rgbdslam_catkin_ws/src/rgbdslam
    linked by target "rgbdslam" in directory /home/jack/Code/rgbdslam_catkin_ws/src/rgbdslam

-- Configuring incomplete, errors occurred!
See also "/home/jack/Code/rgbdslam_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jack/Code/rgbdslam_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

So what's the problem exactly? why i used to install it successfully but failed now? Tanks

felixendres, 2016-12-15:
Somehow g2o is built, but not its cholmod solver. This is confusing. Please provide the output form when g2o is built. Maybe there's a cue as to why that is.

jackj012, 2016-12-15:
@felixendres  Thanks for reply. I have added the output

felixendres, 2016-12-15:
Try `sudo apt-get install libsuitesparse-dev`,
then rebuild g2o and rgbdslam again (either using install.sh or manually)

jackj012, 2016-12-15:
@felixendres  now the cmake process seems good, but another error happens.....
 
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE12solvePatternERNS_17SparseBlockMatrixIS3_EERKSt6vectorISt4pairIiiESaISA_EERKS6_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE12solvePatternERNS_17SparseBlockMatrixIS3_EERKSt6vectorISt4pairIiiESaISA_EERKS6_]+0x67): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0xee): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&, double*, double*)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xf6): undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x1c6): undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE]+0x67): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0xee): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*)':
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xf6): undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
graph_manager.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x1c6): undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
collect2: error: ld returned 1 exit status
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:1544: recipe for target '/home/jack/Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam' failed
make[2]: *** [/home/jack/Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam] Error 1
CMakeFiles/Makefile2:3158: recipe for target 'rgbdslam/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2" failed

felixendres, 2016-12-15:
Oh my. Can you execute `VERBOSE=1 catkin_make -C /home/jack/Code/rgbdslam_catkin_ws/` and post the lines of output that contain "-L". `VERBOSE=1 catkin_make -C /home/jack/Code/rgbdslam_catkin_ws/|grep -- -L` should do that.

ImHos, 2016-12-17:
Hi **felixendres**. I tried to install rgbdslam_v2 using  install.sh as well.
Here is what I got

```
arthur@arthur-ubuntu:~$ VERBOSE=1 catkin_make -C /home/arthur/Code/rgbdslam_catkin_ws/|grep -- -L
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
G2o-libraries /home/arthur/Code/g2ofork/install/lib/libg2o_csparse_extension.so/home/arthur/Code/g2ofork/install/lib/libg2o_core.so/home/arthur/Code/g2ofork/install/lib/libg2o_stuff.so/home/arthur/Code/g2ofork/install/lib/libg2o_types_slam3d.soG2O_SOLVER_CHOLMOD_LIB-NOTFOUND/home/arthur/Code/g2ofork/install/lib/libg2o_solver_pcg.so/home/arthur/Code/g2ofork/install/lib/libg2o_solver_csparse.so
/home/arthur/Code/rgbdslam_catkin_ws/src/rgbdslam/external/SiftGPU/
/home/arthur/Code/rgbdslam_catkin_ws/build/rgbdslam/external/siftgpu_prefix
Using OpenCV from /opt/ros/kinetic/share/OpenCV-3.1.0-dev
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
G2O_SOLVER_CHOLMOD_LIB
    linked by target "rgbdslam" in directory /home/arthur/Code/rgbdslam_catkin_ws/src/rgbdslam
    linked by target "rgbdslam" in directory /home/arthur/Code/rgbdslam_catkin_ws/src/rgbdslam

make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

Thanks

Danmou, 2017-01-27:
@jackj012 Did you resolve your problem? I am getting the exact same errors.

@felixendres Here is the output that you requested (it's all just one line):
`/usr/bin/c++     -fopenmp -O3 -DNDEBUG    CMakeFiles/rgbdslam.dir/src/moc_qtros.cpp.o CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o CMakeFiles/rgbdslam.dir/src/moc_qt_gui.cpp.o CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o CMakeFiles/rgbdslam.dir/src/moc_glviewer.cpp.o CMakeFiles/rgbdslam.dir/src/moc_ros_service_ui.cpp.o CMakeFiles/rgbdslam.dir/src/gicp-fallback.cpp.o CMakeFiles/rgbdslam.dir/src/main.cpp.o CMakeFiles/rgbdslam.dir/src/qtros.cpp.o CMakeFiles/rgbdslam.dir/src/openni_listener.cpp.o CMakeFiles/rgbdslam.dir/src/qt_gui.cpp.o CMakeFiles/rgbdslam.dir/src/node.cpp.o CMakeFiles/rgbdslam.dir/src/glviewer.cpp.o CMakeFiles/rgbdslam.dir/src/parameter_server.cpp.o CMakeFiles/rgbdslam.dir/src/ros_service_ui.cpp.o CMakeFiles/rgbdslam.dir/src/misc.cpp.o CMakeFiles/rgbdslam.dir/src/landmark.cpp.o CMakeFiles/rgbdslam.dir/src/loop_closing.cpp.o CMakeFiles/rgbdslam.dir/src/ColorOctomapServer.cpp.o CMakeFiles/rgbdslam.dir/src/scoped_timer.cpp.o CMakeFiles/rgbdslam.dir/src/icp.cpp.o CMakeFiles/rgbdslam.dir/src/matching_result.cpp.o CMakeFiles/rgbdslam.dir/src/graph_manager.cpp.o CMakeFiles/rgbdslam.dir/src/graph_mgr_io.cpp.o CMakeFiles/rgbdslam.dir/src/graph_manager2.cpp.o CMakeFiles/rgbdslam.dir/src/graph_mgr_odom.cpp.o CMakeFiles/rgbdslam.dir/src/transformation_estimation_euclidean.cpp.o CMakeFiles/rgbdslam.dir/src/transformation_estimation.cpp.o CMakeFiles/rgbdslam.dir/src/header.cpp.o CMakeFiles/rgbdslam.dir/src/covariance_estimation.cpp.o CMakeFiles/rgbdslam.dir/src/feature_adjuster.cpp.o CMakeFiles/rgbdslam.dir/src/features.cpp.o CMakeFiles/rgbdslam.dir/src/sift_gpu_wrapper.cpp.o  -o /home/slam/rgbdslam_v2/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam  -L/opt/ros/kinetic/lib  -L/home/slam/rgbdslam_v2/rgbdslam_catkin_ws/src/lib  -L/home/slam/rgbdslam_v2/rgbdslam_catkin_ws/src/rgbdslam/external/SiftGPU/bin -rdynamic /opt/ros/kinetic/lib/liboctomap.so /opt/ros/kinetic/lib/liboctomath.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_csparse_extension.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_core.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_stuff.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_types_slam3d.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_cholmod.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_pcg.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_csparse.so -lGL -lGLU -lcholmod -lcxsparse /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5.5.1 /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_csparse_extension.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_core.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_stuff.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_types_slam3d.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_cholmod.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_pcg.so /home/slam/rgbdslam_v2/g2ofork/install/lib/libg2o_solver_csparse.so /opt/ros/kinetic/lib/libopencv_xphoto3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_xobjdetect3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_tracking3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_surface_matching3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_structured_light3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_stereo3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_saliency3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_rgbd3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_reg3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_plot3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_optflow3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_line_descriptor3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_hdf3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_fuzzy3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_dpm3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_dnn3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_datasets3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_cvv3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_ccalib3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_bioinspired3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_bgsegm3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_aruco3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_viz3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_videostab3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_superres3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_stitching3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_photo3.so.3.1.0 -lsiftgpu /opt/ros/kinetic/lib/libpcl_ros_filters.so /opt/ros/kinetic/lib/libpcl_ros_io.so /opt/ros/kinetic/lib/libpcl_ros_tf.so -lpcl_common -lpcl_kdtree -lpcl_octree -lpcl_search -lpcl_io -lpcl_sample_consensus -lpcl_filters -lpcl_features -lpcl_segmentation -lpcl_surface -lpcl_registration -lpcl_recognition -lpcl_keypoints -lpcl_visualization -lpcl_people -lpcl_outofcore -lpcl_tracking -lboost_iostreams -lboost_serialization -lqhull -lOpenNI -Wl,-Bstatic -lflann_cpp_s -Wl,-Bdynamic /usr/lib/x86_64-linux-gnu/libvtkImagingStencil-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonComputationalGeometry-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtksys-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonMisc-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonSystem-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkCommonTransforms-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkImagingCore-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu [... cut ...] ux-gnu/libQt5Core.so.5.5.1 /opt/ros/kinetic/lib/libopencv_flann3.so.3.1.0 /opt/ros/kinetic/lib/libopencv_core3.so.3.1.0 -Wl,-rpath,/opt/ros/kinetic/lib:/home/slam/rgbdslam_v2/rgbdslam_catkin_ws/src/lib:/home/slam/rgbdslam_v2/rgbdslam_catkin_ws/src/rgbdslam/external/SiftGPU/bin:/home/slam/rgbdslam_v2/g2ofork/install/lib:/usr/lib/x86_64-linux-gnu/hdf5/serial/lib `

For reference, I am also running Kinetic (on a fairly fresh Ubuntu 16.04 install) but I have not previously had rgbdslam_v2 installed.

Danmou, 2017-01-31:
I have managed to solve the problem: After running `sudo apt-get install libsuitesparse-dev`, it was necessary to remove the two folders generated by install.sh completely and run install.sh again (`make clean` and `catkin_make clean` in the two folders was not sufficient).

closed 2017-03-15 by jackj012