felixendres/rgbdslam_v2 #45: Invoking "make -j8 -l8" failed for rgbdslam_v2

ap108, 2017-01-02:
Hi

When I run catkin_make for rgbdslam_v2 in ROS Indigo, Ubuntu 14.04 LTS. I get the error below. This is only part of the error, couldn't get the rest.

`/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:211:65:   required from ‘Eigen::GeneralProduct<Lhs, Rhs, 3>::GeneralProduct(const Lhs&, const Rhs&) [with Lhs = Eigen::Transpose<Eigen::Matrix<double, 3, 1> >; Rhs = Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >]’
/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:595:60:   required from ‘const typename Eigen::ProductReturnType<Derived, OtherDerived>::Type Eigen::MatrixBase<Derived>::operator*(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >; Derived = Eigen::Transpose<Eigen::Matrix<double, 3, 1> >; typename Eigen::ProductReturnType<Derived, OtherDerived>::Type = Eigen::GeneralProduct<Eigen::Transpose<Eigen::Matrix<double, 3, 1> >, Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >, 3>]’
/home/ap/catkin_ws/src/rgbdslam_v2/src/misc.cpp:763:125:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:56:30: warning: ignoring attributes on template argument ‘Eigen::internal::packet_traits<double>::type {aka __vector(2) double}’ [-Wignored-attributes]
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h: In instantiation of ‘class Eigen::DenseCoeffsBase<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false>, 0>’:
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:566:7:   required from ‘class Eigen::DenseCoeffsBase<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false>, 2>’
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:371:8:   required from ‘struct Eigen::internal::special_scalar_op_base<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false>, double, double, false>’
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:41:34:   required from ‘class Eigen::DenseBase<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> >’
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:48:34:   required from ‘class Eigen::MatrixBase<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> >’
/usr/include/eigen3/Eigen/src/Core/MapBase.h:27:34:   required from ‘class Eigen::MapBase<Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false>, 0>’
/usr/include/eigen3/Eigen/src/Core/Block.h:313:7:   [ skipping 13 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/eigen3/Eigen/src/Core/ReturnByValue.h:61:7:   required from ‘void Eigen::ReturnByValue<Derived>::evalTo(Dest&) const [with Dest = Eigen::Matrix<double, 3, 1>; Derived = Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> >]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:296:7:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::ReturnByValue<OtherDerived>&) [with OtherDerived = Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> >; _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:126:49:   required from ‘Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::CwiseBinaryOp(const Lhs&, const Rhs&, const BinaryOp&) [with BinaryOp = Eigen::internal::scalar_product_op<double, double>; Lhs = const Eigen::Transpose<const Eigen::Transpose<Eigen::Matrix<double, 3, 1> > >; Rhs = const Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >]’
/usr/include/eigen3/Eigen/src/Core/../plugins/MatrixCwiseBinaryOps.h:24:10:   required from ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<typename Eigen::internal::traits<T>::Scalar, typename Eigen::internal::traits<OtherDerived>::Scalar>, const Derived, const OtherDerived> Eigen::MatrixBase<Derived>::cwiseProduct(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >; Derived = Eigen::Transpose<const Eigen::Transpose<Eigen::Matrix<double, 3, 1> > >; typename Eigen::internal::traits<OtherDerived>::Scalar = double; typename Eigen::internal::traits<T>::Scalar = double]’
/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:211:65:   required from ‘Eigen::GeneralProduct<Lhs, Rhs, 3>::GeneralProduct(const Lhs&, const Rhs&) [with Lhs = Eigen::Transpose<Eigen::Matrix<double, 3, 1> >; Rhs = Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >]’
/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:595:60:   required from ‘const typename Eigen::ProductReturnType<Derived, OtherDerived>::Type Eigen::MatrixBase<Derived>::operator*(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >; Derived = Eigen::Transpose<Eigen::Matrix<double, 3, 1> >; typename Eigen::ProductReturnType<Derived, OtherDerived>::Type = Eigen::GeneralProduct<Eigen::Transpose<Eigen::Matrix<double, 3, 1> >, Eigen::ReturnByValue<Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> > >, 3>]’
/home/ap/catkin_ws/src/rgbdslam_v2/src/misc.cpp:763:125:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:56:30: warning: ignoring attributes on template argument ‘Eigen::internal::packet_traits<double>::type {aka __vector(2) double}’ [-Wignored-attributes]
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h: In instantiation of ‘class Eigen::DenseCoeffsBase<Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> >, 0>’:
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:566:7:   required from ‘class Eigen::DenseCoeffsBase<Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> >, 2>’
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:371:8:   required from ‘struct Eigen::internal::special_scalar_op_base<Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> >, double, double, false>’
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:41:34:   required from ‘class Eigen::DenseBase<Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> > >’
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:48:34:   required from ‘class Eigen::MatrixBase<Eigen::Transpose<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false> > >’
/usr/include/eigen3/Eigen/src/Core/Transpose.h:100:37:   required from ‘class Eigen::TransposeImpl<const Eigen::Block<const Eigen::Block<const Eigen::Matrix<double, 3, 3>, 1, 3, false>, 1, 2, false>, Eigen::Dense>’
/usr/include/eigen3/Eigen/src/Core/Transpose.h:57:37:   [ skipping 10 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/eigen3/Eigen/src/Core/ReturnByValue.h:61:7:   required from ‘void Eigen::ReturnByValue<Derived>::evalTo(Dest&) const [with Dest = Eigen::Matrix<double, 3, 1>; Derived = Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> >]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:296:7:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::ReturnByValue<OtherDerived>&) [with OtherDerived = Eigen::internal::solve_retval_base<Eigen::LLT<Eigen::Matrix<double, 3, 3>, 1>, Eigen::Matrix<double, 3, 1> >; _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’
[... cut ...]
make[2]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/src/misc.cpp.o] Error 1
make[1]: *** [rgbdslam_v2/CMakeFiles/rgbdslam.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 10 jobserver tokens available; should be 8!
Invoking "make -j8 -l8" failed`

Here are the [CMakeLists.txt](http://pastebin.com/gxXXaWbL) and  [package.xml](http://pastebin.com/0HmZJ0M9) files.

My pc also freezes sometimes when running catkin_make for this package. I have 8 Gb of RAM and 5 Gb of space left. Is this a memory problem ?

Thank you

felixendres, 2017-01-04:
That is weird. The only error source I can make out is [misc.cpp, line 763](https://github.com/felixendres/rgbdslam_v2/blame/kinetic/src/misc.cpp#L763). Since this didn't change recently, I have no idea what causes this.

Maybe the rest of the error message would help. Please try running `catkin_make -j1` for readability and to exclude the chance of memory exhaustion.

I assume you are working on an x86_64 architecture (see `uname -m`)?

ap108, 2017-01-05:
Thanks for the reply.  I get the output attached when running `catkin_make -j1`. 
Yes, I'm on  x86_64. Thank you
[rgbdslam make _j1.txt](https://github.com/felixendres/rgbdslam_v2/files/687017/rgbdslam.make._j1.txt)

felixendres, 2017-01-09:
Ok, the error is buried deeply:

    /home/ap/catkin_ws/src/rgbdslam_v2/src/misc.cpp:804:25: error: ‘int round(float)’ conflicts with a previous declaration
    /usr/include/c++/6/cmath:1708:3: note: previous declaration ‘constexpr float std::round(float)’

My guess is that, on your system, one of rgbdslam's dependencies includes the namespace "std" in a header file.

Probably the best solution is to explicitly scope the 4 calls of round in misc.cpp to the global scope, by replacing `round(...)` with `::round(...)`. Or rename the function to "round_float_to_int".