felixendres/rgbdslam_v2 #47: CATKIN_MAKE error "make -j4 -l4" failed

adilbonzi, 2017-01-19:
I am witnessing problem in ubuntu 16.04LTS for rgbdslam_v2-hydro. Following the steps posted by Mr. felixendres. I cannot get past this problem showed below. kindly advise me on this matter. This is what is shown after i run the command, "catkin_make".

Base path: /home/adil/rgbdslam_catkin_ws
Source space: /home/adil/rgbdslam_catkin_ws/src
Build space: /home/adil/rgbdslam_catkin_ws/build
Devel space: /home/adil/rgbdslam_catkin_ws/devel
Install space: /home/adil/rgbdslam_catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/adil/rgbdslam_catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/adil/rgbdslam_catkin_ws/build"
####
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[  5%] Built target rgbdslam_generate_messages_lisp
[ 16%] Built target siftgpu_proj
[ 23%] Built target rgbdslam_generate_messages_eus
[ 29%] Built target rgbdslam_generate_messages_cpp
[ 35%] Built target rgbdslam_generate_messages_nodejs
[ 42%] Built target rgbdslam_generate_messages_py
[ 42%] Built target rgbdslam_gencpp
[ 42%] Built target rgbdslam_generate_messages
[ 43%] Generating src/moc_ros_service_ui.cxx
[ 45%] Generating src/moc_qtros.cxx
[ 46%] Generating src/moc_openni_listener.cxx
[ 47%] Generating src/moc_qt_gui.cxx
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/build.make:62: recipe for target 'rgbdslam_v2-hydro/src/moc_qtros.cxx' failed
make[2]: *** [rgbdslam_v2-hydro/src/moc_qtros.cxx] Error 1
make[2]: *** Waiting for unfinished jobs....
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/build.make:87: recipe for target 'rgbdslam_v2-hydro/src/moc_ros_service_ui.cxx' failed
make[2]: *** [rgbdslam_v2-hydro/src/moc_ros_service_ui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/build.make:72: recipe for target 'rgbdslam_v2-hydro/src/moc_qt_gui.cxx' failed
make[2]: *** [rgbdslam_v2-hydro/src/moc_qt_gui.cxx] Error 1
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/build.make:67: recipe for target 'rgbdslam_v2-hydro/src/moc_openni_listener.cxx' failed
make[2]: *** [rgbdslam_v2-hydro/src/moc_openni_listener.cxx] Error 1
CMakeFiles/Makefile2:3158: recipe for target 'rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam_v2-hydro/CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

felixendres, 2017-01-20:
ROS Hydro is meant for 12.04. Which ROS do you have? Try the kinetic branch. Make sure to read the installation instructions in the README.md of that branch, as it differs from previous versions.

adilbonzi, 2017-01-21:
I installed ROS hydro and used the kinetic branch. I followed all the instructions accordingly as stated in the README.md. Kindly suggest which branch to use for ubuntu 16.04.

felixendres, 2017-02-15:
Sorry for the late reply. Your ubuntu and ROS need to be compatible. [Hydro does not support ubuntu 16.04](http://www.ros.org/reps/rep-0003.html#hydro-medusa-aug-2013). The solution therefore would be to switch from ROS hydro to ROS kinetic.

closed 2017-02-15 by felixendres

1401609738, 2019-08-17:
fuck you !You the idiot. You didn't even give an available solution.