felixendres/rgbdslam_v2 #49: Rgb color doesn't match point cloud

Hannesiii, 2017-03-01:
Hello,
I am using the rgbdslam_v2 with a Kinect_v2 on ROS indigo (iai_kinect2 +freenect) to create a point cloud from an object (moving on a circular path around the object, Kinect attached to an universal robot arm). But somehow the rgb image doesn't match the point cloud, it seems that there is a kind of offset between them. Distance btw. object and sensor is around 0,6m. Does anybody know what could be the reason for this behavior?
Thank you in advance!
Hannes

closed 2017-03-02 by Hannesiii

DalderupMaurice, 2017-03-28:
How exactly did you calibrate? @Hannesiii

Hannesiii, 2017-04-11:
@DalderupMaurice 
Look here: https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration

DalderupMaurice, 2017-04-12:
Thanks @Hannesiii ! and no problem, I already got it working but not very accurate. I might have to re-calibrate, thanks. 

One last question though. Whenever I use rgbd, the performance is extremely slow and it updates realy slow. Do you have any clue on what the cause of this is? I am new to SLAM and have no idea about what causes the slow performance and what the parameters in the launch file can do.