felixendres/rgbdslam_v2 #53: Build errors about qt5

xiteen, 2017-05-19:
Hello,
There are some errors about qt5 when I built this package.
I try to install qt5 but it does not work. 
Please tell me how to fix it.

dev@ubuntu:~/catkin_ws$ catkin_make
Base path: /home/dev/catkin_ws
Source space: /home/dev/catkin_ws/src
Build space: /home/dev/catkin_ws/build
Devel space: /home/dev/catkin_ws/devel
Install space: /home/dev/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/dev/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/dev/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dev/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rgbdslam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rgbdslam'
-- ==> add_subdirectory(rgbdslam_v2)
-- SiftGPU library will be used
-- Using these message generators: gencpp;genlisp;genpy
-- rgbdslam: 0 messages, 4 services
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- Searching for g2o ...
-- Found g2o headers in: /opt/ros/indigo/include
-- Found libg2o: /opt/ros/indigo/lib/libg2o_csparse_extension.so;/opt/ros/indigo/lib/libg2o_core.so;/opt/ros/indigo/lib/libg2o_stuff.so;/opt/ros/indigo/lib/libg2o_types_slam3d.so;/opt/ros/indigo/lib/libg2o_solver_cholmod.so;/opt/ros/indigo/lib/libg2o_solver_pcg.so;/opt/ros/indigo/lib/libg2o_solver_csparse.so
G2o-libraries /opt/ros/indigo/lib/libg2o_csparse_extension.so/opt/ros/indigo/lib/libg2o_core.so/opt/ros/indigo/lib/libg2o_stuff.so/opt/ros/indigo/lib/libg2o_types_slam3d.so/opt/ros/indigo/lib/libg2o_solver_cholmod.so/opt/ros/indigo/lib/libg2o_solver_pcg.so/opt/ros/indigo/lib/libg2o_solver_csparse.so
CMake Warning at rgbdslam_v2/CMakeLists.txt:98 (find_package):
  By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.

CMake Warning at rgbdslam_v2/CMakeLists.txt:99 (find_package):
  By not providing "FindQt5OpenGL.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5OpenGL", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5OpenGL" with
  any of the following names:

    Qt5OpenGLConfig.cmake
    qt5opengl-config.cmake

  Add the installation prefix of "Qt5OpenGL" to CMAKE_PREFIX_PATH or set
  "Qt5OpenGL_DIR" to a directory containing one of the above files.  If
  "Qt5OpenGL" provides a separate development package or SDK, be sure it has
  been installed.

CMake Warning at rgbdslam_v2/CMakeLists.txt:100 (find_package):
  By not providing "FindQt5Concurrent.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "Qt5Concurrent", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Concurrent"
  with any of the following names:

    Qt5ConcurrentConfig.cmake
    qt5concurrent-config.cmake

  Add the installation prefix of "Qt5Concurrent" to CMAKE_PREFIX_PATH or set
  "Qt5Concurrent_DIR" to a directory containing one of the above files.  If
  "Qt5Concurrent" provides a separate development package or SDK, be sure it
  has been installed.

/home/dev/catkin_ws/src/rgbdslam_v2/external/SiftGPU/
/home/dev/catkin_ws/build/rgbdslam_v2/external/siftgpu_prefix
Using OpenCV from /usr/share/OpenCV
CMake Error at rgbdslam_v2/CMakeLists.txt:212 (qt5_add_resources):
  Unknown CMake command "qt5_add_resources".

-- Configuring incomplete, errors occurred!
See also "/home/dev/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dev/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

felixendres, 2017-05-20:
Maybe you need the -dev packages of qt5. What did you install?

xiteen, 2017-05-21:
I need the indigo branch to build. But I git clone the kinetic branch. So the problem is coming.

closed 2017-05-21 by xiteen