felixendres/rgbdslam_v2 #59: No data in Octomap

robotsoft, 2017-06-05:
I've tried to save a octomap but there is no data in an octomap file. I used 'Save/Save Octomap..." in the menu to save the octomap. The generated octomap includes just headers as follows.

```
# Octomap OcTree file
# (feel free to add / change comments, but leave the first line as it is!)
#
id ColorOcTree
size 0
res 0.05
data
```
I'm working on Ubuntu 16.04. Is there any I missed to save the octomap?

Thank you.

robotsoft, 2017-06-05:
I solved this issue by adding `return true` in bool GraphManager::updateCloudOrigin(Node* node). This function needs to return a bool but 'return true' is missed in the end of the function. The function can be found in graph_mgr_io.cpp. I have finally a binary octomap file.

zhangduoshou, 2017-07-12:
@robotsoft hello，could u paste how u modified the code.I met the same problem with u and not solved until

zhangduoshou, 2017-07-12:
@robotsoft I solved the problem with the method u metioned，thank u very much