felixendres/rgbdslam_v2 #60: REQUIRED process [rgbdslam-1] has died!

AmirCognitive, 2017-06-06:
when I run the roslaunch command for only rgbdslam I get the following error, your help will be appreciated.

amir-ai@amir-comp:~$ roslaunch rgbdslam rgbdslam.launch 
... logging to /home/amir-ai/.ros/log/03a9fc06-4a90-11e7-936b-38d547df45ba/roslaunch-amir-comp-23778.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://amir-comp:39035/

SUMMARY
========

PARAMETERS
 * /rgbdslam/config/backend_solver: pcg
 * /rgbdslam/config/cloud_creation_skip_step: 2
 * /rgbdslam/config/cloud_display_type: POINTS
 * /rgbdslam/config/detector_grid_resolution: 3
 * /rgbdslam/config/feature_detector_type: ORB
 * /rgbdslam/config/feature_extractor_type: ORB
 * /rgbdslam/config/max_keypoints: 600
 * /rgbdslam/config/max_matches: 300
 * /rgbdslam/config/min_sampled_candidates: 4
 * /rgbdslam/config/neighbor_candidates: 4
 * /rgbdslam/config/optimizer_skip_step: 1
 * /rgbdslam/config/pose_relative_to: largest_loop
 * /rgbdslam/config/predecessor_candidates: 4
 * /rgbdslam/config/ransac_iterations: 100
 * /rgbdslam/config/topic_image_depth: /camera/depth_reg...
 * /rgbdslam/config/topic_image_mono: /camera/rgb/image...
 * /rgbdslam/config/topic_points: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    rgbdslam (rgbdslam/rgbdslam)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rgbdslam-1]: started with pid [23796]
================================================================================REQUIRED process [rgbdslam-1] has died!
process has died [pid 23796, exit code -11, cmd /home/amir-ai/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/amir-ai/.ros/log/03a9fc06-4a90-11e7-936b-38d547df45ba/rgbdslam-1.log].
log file: /home/amir-ai/.ros/log/03a9fc06-4a90-11e7-936b-38d547df45ba/rgbdslam-1*.log
Initiating shutdown!
================================================================================
[rgbdslam-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MichaelCZhou, 2017-06-10:
i have the same question with you...

AmirCognitive, 2017-06-10:
Michael, my problem solved after I used the indigo branch, and you have to check the camera and depth topics as well, and also in my case it does not work with openni2 but it works with kinect.

MichaelCZhou, 2017-06-10:
you mean you use indigo branch in ubuntu16.04?thank you

AmirCognitive, 2017-06-10:
No, in ubuntu 16 there was other issues, so I ran it in ubuntu 14, opencv 2.4.11, ros-indigo-opencv3, ros indigo. Welcome.

chriswsuarez, 2017-07-28:
Are there any other solutions to this? I have to be on 16.04 and kinetic but I am getting the same error when trying to launch rgbdslam.launch.

SILI1994, 2017-11-14:
has anyone solved this problem? facing the same problem now

songlin, 2018-01-05:
Having the same problem here with Ros Kinetic, even tryied the kinetic branch.

MariRob, 2018-03-08:
hi !
i am trying to get it working with Orbbec Astra on Indigo 14.04
i got this 
[image]!

how can i solve this ? any suggestion ?

sararht, 2018-08-07:
I have the same problem as you @canberkgurel  . Did you fix the problem? Thank you.

KamithaVihanga, 2018-09-11:
The new release of rgbdslam_v2 doesn't support previous drivers. We just solved this by cloning older version of rgbdslam_v2 in github.

hirunisilva275, 2018-09-11:
@KamithaVihanga  thank you. That was helpful. I had the same problem and I could solve it after using this method.

arunumd, 2018-12-07:
@KamithaVihanga and @hirunisilva275  I am still facing the same problem as shown below:
`================================================================================REQUIRED process [rgbdslam-35] has died!
process has died [pid 27215, exit code -11, cmd /home/arun/Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam /camera/rgb/image_color:=/camera/rgb/image_raw /camera/depth_registered/sw_registered/image_rect_raw:=/camera/depth/image_raw __name:=rgbdslam __log:=/home/arun/.ros/log/63602fc8-fa5f-11e8-86d6-5cea1d834771/rgbdslam-35.log].
log file: /home/arun/.ros/log/63602fc8-fa5f-11e8-86d6-5cea1d834771/rgbdslam-35*.log
Initiating shutdown!
================================================================================
`
I see you have found a workaround for this by installing an older version. Can you please guide me which version solved this problem ?

KamithaVihanga, 2018-12-10:
@arunumd I have used the file which was cloned earlier. Did you fixed that one?

arunumd, 2018-12-11:
@KamithaVihanga  I tried a previous commit. But it still gave me the same error. So I gave up with rgbdslam and moved on. Would be interested to know if it works for anyone else.... I would like to use it later !

yuhiuyang, 2019-02-12:
@songlin what version of pcl do you use? pcl 1.8 or pcl 1.7

florianspy, 2019-12-03:
To those who had problems "REQUIRED process [rgbdslam-1] has died!" with following the guide on https://github.com/felixendres/rgbdslam_v2/wiki/Instructions-for-Compiling-Rgbdslam-(V2)-on-a-Fresh-Ubuntu-16.04-Install-(Ros-Kinetic)-in-Virtualbox especially steps 11-16 with pcl setup and recompiling solved my problem.