felixendres/rgbdslam_v2 #64: g2o compile error

adnion, 2017-06-20:
Using Ubuntu 16.04 I can't compile the g2o fork. 
```
[ 34%] Building CXX object g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2_pointxy_bearing.cpp.o

catkin_ws/src/g2o/g2o/types/slam2d/edge_se2_pointxy_bearing.cpp: In member function ‘virtual void g2o::EdgeSE2PointXYBearing::initialEstimate(const VertexSet&, g2o::OptimizableGraph::Vertex*)’:
catkin_ws/src/g2o/g2o/types/slam2d/edge_se2_pointxy_bearing.cpp:51:52: error: no matching function for call to ‘g2o::SE2::setRotation(Eigen::Rotation2D<double>::Scalar)’
     t.setRotation(t.rotation().angle()+_measurement);
                                                    ^
In file included from catkin_ws/src/g2o/g2o/types/slam2d/vertex_se2.h:33:0,
                 from catkin_ws/src/g2o/g2o/types/slam2d/edge_se2_pointxy_bearing.h:31,
                 from catkin_ws/src/g2o/g2o/types/slam2d/edge_se2_pointxy_bearing.cpp:27:
catkin_ws/src/g2o/g2o/types/slam2d/se2.h:59:12: note: candidate: void g2o::SE2::setRotation(const Rotation2Dd&)
       void setRotation(const Eigen::Rotation2Dd& R_) {_R=R_;}
            ^
catkin_ws/src/g2o/g2o/types/slam2d/se2.h:59:12: note:   no known conversion for argument 1 from ‘Eigen::Rotation2D<double>::Scalar {aka double}’ to ‘const Rotation2Dd& {aka const Eigen::Rotation2D<double>&}’
g2o/types/slam2d/CMakeFiles/types_slam2d.dir/build.make:206: recipe for target 'g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2_pointxy_bearing.cpp.o' failed
make[2]: *** [g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2_pointxy_bearing.cpp.o] Error 1
CMakeFiles/Makefile2:1016: recipe for target 'g2o/types/slam2d/CMakeFiles/types_slam2d.dir/all' failed
make[1]: *** [g2o/types/slam2d/CMakeFiles/types_slam2d.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

adnion, 2017-06-24:
Tried the installation script. It fails with:
`/usr/local/include/g2o/core/robust_kernel_impl.h:60:7: error: ‘RobustKernelPtr’ does not name a type
       RobustKernelPtr _kernel;
       ^
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:158: recipe for target 'rgbdslam/CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o' failed
make[2]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/src/moc_graph_manager.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:110: recipe for target 'rgbdslam/CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o' failed
make[2]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/src/moc_openni_listener.cpp.o] Error 1
CMakeFiles/Makefile2:3038: recipe for target 'rgbdslam/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2" failed
`

blueeaglex, 2017-07-17:
Hi. Have you sloved it now?

adnion, 2017-07-17:
Haven't tried it again...

blueeaglex, 2017-07-18:
@adnion 
You may miss some dependenceies.

blueeaglex, 2017-07-18:
@adnion  See my pull request #65