felixendres/rgbdslam_v2 #66: What kind of changs required to make it work with Orbbec Astra

mzaiady, 2017-07-18:
Trying to get it working with Orbbec Astra on Indigo, but it's always giving "waiting for image on topic", although I can visualize  the topic on rviz!

geret13, 2017-09-26:
+1 on kinetic

Blennes77, 2018-01-04:
Hello and happy new year !
Like mzaiady and geret13 , could you provide us some feedback to make working rgbdslam_v2 with orbbec astra cam ?

Regards.

MariRob, 2018-03-06:
Hi ! 
I have the same prblm , did u solve it ,if yes , can you show me how 
Thakns

Blennes77, 2018-03-06:
Hi MariRob !
I finished to make it work. I hope that will work for you too. In the file "rgbdslam.launch" you'll find a parameter named "config/topic_image_mono" which is bind on the topic "/camera/rgb/image_rect_color ".
Modify this topic by "/camera/rgb/image_color". It's worked for me.

Regards.

Blennes77, 2018-03-06:
after execute those commands : 
roslaunch turtlebot_bringup miminal.launch
rosrun turtlebot_teleop teleop_keyboard.launch (if you want the turtlebot move)
roslaunch astra_launch astra.launch (for the Orbbec Astra)
rosrun rgbdslam rgbdslam.launch

MariRob, 2018-03-07:
thank you so much for your reply @Blennes77  ^^
but  the parameter  "config/topic_image_mono" is originally bind on the topic "/camera/rgb/image_color"
I'am trying to keep the same name of topic "camera/depth_regisetred/sw_registered/image_rect_raw" which wait rgbdslam  and change the name of topic send by the camera 
I think they haven't the same name.
have you any other solution ? 
[image]

Blennes77, 2018-03-08:
Hi MariRob,

I'm confuse, I've told to you exactly the opposite that i wanted say ! You must replace the topic "/camera/rgb/image_color" by "/camera/rgb/image_rect_color".
[image]

MariRob, 2018-03-08:
hello @Blennes77 
already done but without success...
I really don't know where the problem is :( 
[image]
[image]

Blennes77, 2018-03-08:
Did you launch this before ? "roslaunch astra_launch astra.launch"

MariRob, 2018-03-08:
yes i did 
no data ...  
[image]

Blennes77, 2018-03-08:
So your Astra doesn't send his data to this topic.

MariRob, 2018-03-08:
do you know why? 
I really thank you @Blennes77 . I appreciate <3

Blennes77, 2018-03-08:
You could test with all of the camera topics in rviz by adding a camera object and selecting a video topic in the dropdown list. When the video appear you replace the topic in rgbdslam.launch.

MariRob, 2018-03-14:
Hi again @Blennes77 
I wanna ask you about rgbdslam ( still not working :(((  )
does rgbdslam execute only if **visual image , depth image and feature image** are launched together ? because i have visual and depth image but i have no idea how to get  "feature image" 
"Waiting for feature image.."   what does it mean ?? 
can you help me plz 
Thanks :)