felixendres/rgbdslam_v2 #67: Is there an option to use ORB_GPU in rgbdslam_v2?

blueeaglex, 2017-07-20:
There is no embedded computer with a strong CPU such as i7-4790 for now. But I got Jetson TK1 [http://elinux.org/Jetson_TK1](url)
So I would like to try ORB_GPU in rgbdslam_V2. 
I read some documentation of some other project such as ORB_SLAM2 [https://github.com/raulmur/ORB_SLAM2](url) . But this option was not even metioned!!!!
I just want to know why ORB_CPU is so unpopular in area of SLAM.
Any feedback will be greatly appreciated

felixendres, 2017-07-20:
When this project started there was no embedded computer with a strong GPU. If you can make ORB_GPU work, I'll gladly accept a pull request.

blueeaglex, 2017-07-21:
@felixendres  OK, I see. I'll try this sooner or later.
Thanks.