felixendres/rgbdslam_v2 #68: How to refining the registration by ICP?

blueeaglex, 2017-07-20:
The only information I can get is "GICP Generalized ICP can be (de)activated for refining the registration. For more information see http://stanford.edu/~avsegal/generalized_icp.html"
But the url is not accessable.
How can I get the registered pointcloud sequence rather than a merged large single pointcloud(The quicksave button)?

felixendres, 2017-07-20:
Using ICP mainly was an experiment by a co-author. Since it didn't improve results consistently in acceptable time it hasn't been used for a/usr/lib/libcxsparse.so.3.1.2 loong time and I do not recommend using it. For some results, see [Tabelle 2 in this (German) publication](http://www2.informatik.uni-freiburg.de/~endres/files/publications/endres12at.pdf).

For the sequence of clouds, there's the option "Save Point Cloud Node-Wise..." in the "Save" menu. Or you can save as a bag file, then you get the original clouds plus the transformations in tf messages.

blueeaglex, 2017-07-21:
@felixendres 
Thank you very much
Although I can not read German, I can see the results. Yes,the result is not reasonable. I also tested ICP registration 1 years ago.  It's not as good as the rumors. But by combining point-to-point ICP，point-to-plane ICP and lum, It could be much better. Of course, this is just a simple combination, and I ignored 
the cost of time.  In other words，I don't care about time， I just want good results. But even then, it was
still not good enough. rgbdslam_v2 did much better.
Well. Now I can get the sequence, and I will be able to restart my experiment on a sequence of registered
point cloud. I'll share the results if I get something reasonable.
Thanks again.