felixendres/rgbdslam_v2 #74: ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies: Missing resource pcl

haidela, 2017-10-14:
Good morning everyone, I am trying to evaluate RGBD-SLAM with our datasets RGB-D SLAM and Benchmark in ROS indigo + Ubuntu 14.04.5 Desktop am64 under VMware workstation 12 Pro, when I run rosdep: ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies
I red some issues on line and did them no result

haidara@haidara-virtual-machine:~/catkin_ws$ rosdep install rgbdslam_eval
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rgbdslam_eval: Missing resource pcl
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/haidara/catkin_ws/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks

Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haidara-virtual-machine:33677/
ros_comm version 1.11.21

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES

auto-starting new master
process[master]: started with pid [114446]
ROS_MASTER_URI=http://haidara-virtual-machine:11311/

setting /run_id to 78d8c876-b085-11e7-ab99-000c290ff4ea
process[rosout-1]: started with pid [114459]
started core service [/rosout]

Anyone who has ideas about this would be appreciated.

felixendres, 2017-10-15:
Could you please verify that you checked out the indigo branch of rgbdslam?

haidela, 2017-10-17:
@felixendres Thank you for your reply, and I am so sorry that I responded so late to your message, I will try it today and I'll tell you if it works or not.

haidela, 2017-10-23:
@felixendres thank you, it works now

closed 2017-11-03 by haidela