felixendres/rgbdslam_v2 #76: getting compile error g20

rnunziata, 2017-10-29:
getting compile error g20....any idea on how to resolve?

```
/home/rjn/catkin_ws/src/rgbdslam/Thirdparty/g2o/g2o/apps/g2o_viewer/main_window.cpp:378:23: error: ‘ProjType’ is not a class or namespace
   ProjType t = flag ? ProjType::ORTHOGRAPHIC : ProjType::PERSPECTIVE;
```

rnunziata, 2017-10-29:
this worked as a quick get around:
```
  //ProjType t = flag ? ProjType::ORTHOGRAPHIC : ProjType::PERSPECTIVE;
  viewer->camera()->setType(qglviewer::Camera::ORTHOGRAPHIC);
```

rnunziata, 2017-10-29:
This resulted other errors....I will give up for now.

```
CMakeFiles/create_sphere.dir/create_sphere.cpp.o: In function `main':
create_sphere.cpp:(.text.startup+0x1c2b): undefined reference to `g2o::sampleGaussian(std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>*)'
create_sphere.cpp:(.text.startup+0x1de6): undefined reference to `g2o::sampleGaussian(std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>*)'
collect2: error: ld returned 1 exit status
g2o/examples/sphere/CMakeFiles/create_sphere.dir/build.make:101: recipe for target '../bin/create_sphere' failed
make[2]: *** [../bin/create_sphere] Error 1
CMakeFiles/Makefile2:1999: recipe for target 'g2o/examples/sphere/CMakeFiles/create_sphere.dir/all' failed
make[1]: *** [g2o/examples/sphere/CMakeFiles/create_sphere.dir/all] Error 2

```

rnunziata, 2017-10-29:
closing: did not follow instr exactly , needed to compile as
cmake .. -DCMAKE_INSTALL_PREFIX=$G2O_REPO_DIR/install -DG2O_BUILD_EXAMPLES=OFF

closed 2017-10-29 by rnunziata

abdur4373, 2019-02-25:
Hello @rnunziata!

I am also getting same [error](https://github.com/felixendres/rgbdslam_v2/issues/110). I have tried multiple times building g2o library from scratch. Using 
cmake .. -DCMAKE_INSTALL_PREFIX=$G2O_REPO_DIR/install -DG2O_BUILD_EXAMPLES=OFF
but still getting the same error as you have described. Any chance of doing anything extra to resolve the problem.

rnunziata, 2019-02-25:
sorry I no longer have this project installed. Other then making sure you the the system is pointing at the new G2O environment don't know what else I have have done to get this to work.