felixendres/rgbdslam_v2 #84: Error while evaluating rgbslam_v2 with tum benchmark

muhammadaly, 2018-01-26:
I received the following errors while evaluating rgbdslam_v2 package with TUM benchmark 

rgbd_dataset_freiburg1_desk2 ... No estimate at level  0

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 0 are stored in /home/muhammad/ros_workspaces/rgbdslam/src/rgbdslam_v2/test/test1/emm__0.00/CANDIDATES_4/RANSAC_100/OPT_SKIP_10/ORB/600_Features/maxate_evaluation_0.csv
600 Features      "ATE MAX"        "Duration"        "Optimization Time"        Nodes  Edges
rgbd_dataset_freiburg1_desk2 ... No estimate at level  1

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 1 are stored in /home/muhammad/ros_workspaces/rgbdslam/src/rgbdslam_v2/test/test1/emm__0.00/CANDIDATES_4/RANSAC_100/OPT_SKIP_10/ORB/600_Features/maxate_evaluation_1.csv
600 Features      "ATE MAX"        "Duration"        "Optimization Time"        Nodes  Edges
rgbd_dataset_freiburg1_desk2 ... No estimate at level  2

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 2 are stored in /home/muhammad/ros_workspaces/rgbdslam/src/rgbdslam_v2/test/test1/emm__0.00/CANDIDATES_4/RANSAC_100/OPT_SKIP_10/ORB/600_Features/maxate_evaluation_2.csv
600 Features      "ATE MAX"        "Duration"        "Optimization Time"        Nodes  Edges
rgbd_dataset_freiburg1_desk2 ... No estimate at level  3

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 3 are stored in /home/muhammad/ros_workspaces/rgbdslam/src/rgbdslam_v2/test/test1/emm__0.00/CANDIDATES_4/RANSAC_100/OPT_SKIP_10/ORB/600_Features/maxate_evaluation_3.csv
600 Features      "ATE MAX"        "Duration"        "Optimization Time"        Nodes  Edges
rgbd_dataset_freiburg1_desk2 ... No estimate at level  4

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 4 are stored in /home/muhammad/ros_workspaces/rgbdslam/src/rgbdslam_v2/test/test1/emm__0.00/CANDIDATES_4/RANSAC_100/OPT_SKIP_10/ORB/600_Features/maxate_evaluation_4.csv
600 Features      "ATE MAX"        "Duration"        "Optimization Time"        Nodes  Edges

felixendres, 2018-07-09:
The output files of the evaluation script are missing. Since you don't state what you did and how to reproduce the error, I can't really help you there.

OwlCoder, 2018-11-23:
Hi Felix,

I am getting the same error as reported above. 

After building rgbdslam in my catkin_workspace, I installed one of the dataset (rgbd_dataset_freiburg1_360.bag,rgbd_dataset_freiburg1_360-groundtruth.txt) in the folder named benchmark_data.

On running the following command:
roscore &
rosrun rgbdslam benchmark_evaluation.sh SURF /home/naman/catkin_ws/src/rgbdslam_v2/rgbd_benchmark/benchmark_data/

`Evaluating SIFTGPU
Will evaluate RGBD-SLAM on the following bagfiles:
rgbd_dataset_freiburg1_360.bag
19:04:55 Results for rgbd_dataset_freiburg1_360 are stored in /home/naman/catkin_ws/src/rgbdslam_v2/rgbd_benchmark/benchmark_data/emm__0.00/CANDIDATES_8/RANSAC_300/HellingerDistance_true/NN_0.90/MIN_MATCHES_020/SIFTGPU/1200_Features/rgbd_dataset_freiburg1_360
19:04:59 Finished processing rgbd_dataset_freiburg1_360
mv: cannot stat 'rgbd_dataset_freiburg1_360.bag?*': No such file or directory
rgbd_dataset_freiburg1_360 ... No estimate at level  0

paste: eval_translational.ate.txt: No such file or directory
ATE Results at Level 0 are stored in /home/naman/catkin_ws/src/rgbdslam_v2/rgbd_benchmark/benchmark_data/emm__0.00/CANDIDATES_8/RANSAC_300/HellingerDistance_true/NN_0.90/MIN_MATCHES_020/SIFTGPU/1200_Features/maxate_evaluation_0.csv
`