felixendres/rgbdslam_v2 #99: REQUIRED process [rgbdslam-2] has died!

sararht, 2018-08-07:
I am working with Ubuntu 16.04 and ROS Kinetic. 

The installation of rgbdslam_v2 was succesful, but when I execute the command: $ _roslaunch rgbdslam rgbdslam.launch_    i get the following error:

**REQUIRED process [rgbdslam-2] has died!
process has died [pid 31396, exit code -11, cmd /home/sara/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/sara/.ros/log/2afc3b50-9a41-11e8-a6a1-68a86d2b7bba/rgbdslam-2.log].
log file: /home/sara/.ros/log/2afc3b50-9a41-11e8-a6a1-68a86d2b7bba/rgbdslam-2*.log
Initiating shutdown!**

How can I fix this? 
Thank you.

arnabsinha, 2018-09-24:
I am also getting the same problem

chataign, 2018-10-03:
I am also getting a segmentation fault, and this is the output from gdb:

```
Starting program: /home/fchataigner/catkin_ws/devel/lib/rgbdslam/rgbdslam 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5c6e2d4 in long double boost::math::lanczos::lanczos17m64::lanczos_sum<long double>(long double const&) ()
   from /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.7
(gdb) bt
#0  0x00007ffff5c6e2d4 in long double boost::math::lanczos::lanczos17m64::lanczos_sum<long double>(long double const&) ()
   from /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.7
#1  0x00007ffff5bd9d7f in ?? () from /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.7
#2  0x00007ffff7de76ba in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffd028, env=env@entry=0x7fffffffd038)
    at dl-init.c:72
#3  0x00007ffff7de77cb in call_init (env=0x7fffffffd038, argv=0x7fffffffd028, argc=1, l=<optimized out>) at dl-init.c:30
#4  _dl_init (main_map=0x7ffff7ffe168, argc=1, argv=0x7fffffffd028, env=0x7fffffffd038) at dl-init.c:120
#5  0x00007ffff7dd7c6a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#6  0x0000000000000001 in ?? ()
#7  0x00007fffffffd4ef in ?? ()
#8  0x0000000000000000 in ?? ()
(gdb) 

```

chataign, 2018-10-03:
see https://github.com/felixendres/rgbdslam_v2/issues/8

xurz97, 2018-11-14:
Same here. I fix it with [a blog](https://hubpages.com/technology/Instructions-for-compiling-rgbdslam-v2-on-a-fresh-Ubuntu-1604-install-ros-kinetic-in-virtualbox).
Maybe your current version of g2o and pcl is insufficient for rgbd_slam_v2. Unzip eigen3.2.10(no need to install) and reinstall g2o and pcl1.8 may be helpful.