felixendres/rgbdslam_v2 #100: Ubuntu 18.04 and ROS Melodic.

a4zhangfei, 2018-11-10:
Base path: /home/zhangzhihao/rgbdslam_catkin_ws
Source space: /home/zhangzhihao/rgbdslam_catkin_ws/src
Build space: /home/zhangzhihao/rgbdslam_catkin_ws/build
Devel space: /home/zhangzhihao/rgbdslam_catkin_ws/devel
Install space: /home/zhangzhihao/rgbdslam_catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/zhangzhihao/rgbdslam_catkin_ws/build"
####
####
#### Running command: "make -j4" in "/home/zhangzhihao/rgbdslam_catkin_ws/build"
####
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
[  0%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[ 11%] Built target siftgpu_proj
[ 23%] Built target rgbdslam_generate_messages_eus
[ 23%] Built target rgbdslam_generate_messages_nodejs
[ 35%] Built target rgbdslam_generate_messages_cpp
[ 36%] Built target rgbdslam_generate_messages_py
[ 42%] Built target rgbdslam_generate_messages_lisp
[ 42%] Built target rgbdslam_gencpp
[ 46%] Generating src/moc_qtros.cxx
[ 46%] Generating src/moc_openni_listener.cxx
[ 46%] Generating src/moc_ros_service_ui.cxx
[ 46%] Built target rgbdslam_generate_messages
/usr/include/boost/predef/language/stdc.h:52: Parse error at "defined"
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:62: recipe for target 'rgbdslam/src/moc_qtros.cxx' failed
make[2]: *** [rgbdslam/src/moc_qtros.cxx] Error 1
make[2]: *** waiting jobs unfinished....
[ 47%] Generating src/moc_qt_gui.cxx
/usr/include/boost/predef/language/stdc.h:52: Parse error at "defined"
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:67: recipe for target 'rgbdslam/src/moc_openni_listener.cxx' failed
make[2]: *** [rgbdslam/src/moc_openni_listener.cxx] Error 1
/usr/include/boost/predef/language/stdc.h:52: Parse error at "defined"
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:87: recipe for target 'rgbdslam/src/moc_ros_service_ui.cxx' failed
make[2]: *** [rgbdslam/src/moc_ros_service_ui.cxx] Error 1
/usr/include/boost/predef/language/stdc.h:52: Parse error at "defined"
rgbdslam/CMakeFiles/rgbdslam.dir/build.make:72: recipe for target 'rgbdslam/src/moc_qt_gui.cxx' failed
make[2]: *** [rgbdslam/src/moc_qt_gui.cxx] Error 1
CMakeFiles/Makefile2:531: recipe for target 'rgbdslam/CMakeFiles/rgbdslam.dir/all' failed
make[1]: *** [rgbdslam/CMakeFiles/rgbdslam.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Magical-E, 2019-07-24:
Hello,I meet the same error in Ubuntu 18.04 .Have you already solved it?

a4zhangfei, 2019-07-24:
I changed the Ubuntu version: 16.04

ideechaniz, 2019-11-26:
I have the same problem under debian 10 and ros melodic

QianYC, 2019-12-25:
there is an inline function named "round" in misc.cpp, you can solve this problem by simply rename the function and the revoke to anything like "myround"