felixendres/rgbdslam_v2 #104: Error upon roslaunch rgbdslam rgbdslam.launch

joshuayuan, 2018-12-10:
When I run `roslaunch rgbdslam rgbdslam.launch`, I get the following errors after the roslaunch server starts up. Then it dies. The instructions [here](https://hubpages.com/technology/Instructions-for-compiling-rgbdslam-v2-on-a-fresh-Ubuntu-1604-install-ros-kinetic-in-virtualbox) are too complicated for me and feel unnecessary, and are a bit out dated. Hopefully there's some other fix. 

```process[rgbdslam-1]: started with pid [29367]
================================================================================REQUIRED process [rgbdslam-1] has died!
process has died [pid 29367, exit code -11, cmd /home/aaron/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/aaron/.ros/log/6b13f86a-fc4b-11e8-9960-f0421c1d4ccd/rgbdslam-1.log].
log file: /home/aaron/.ros/log/6b13f86a-fc4b-11e8-9960-f0421c1d4ccd/rgbdslam-1*.log
Initiating shutdown!
================================================================================
[rgbdslam-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```
Any ideas, thanks!

WeidiDeng, 2018-12-11:
comment line “add_compile_options(-std=c++11)” in CMakeLists.txt, 
[commit](https://github.com/felixendres/rgbdslam_v2/pull/97) breaks rgbdslam

rmartins86, 2019-01-18:
I had the same issue and @WeidiDeng answer solve it for me :-)

dhiegomaga, 2019-02-21:
Had same issue.

> comment the line “add_compile_options(-std=c++11)” in CMakeLists.txt,
> [this commit](https://github.com/felixendres/rgbdslam_v2/pull/97) breaks rgbdslam, the process segment faults at start because pcl has to be compiled with c++11

Ok, but then it breaks `qt_gui.cpp.o` and a lot of other stuff that needs C++11... 

`/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.`

...

And what you said doesn't make sense to me, if pcl has to be compiled with C++11, how would disable C++11 compilation help?

dhiegomaga, 2019-02-22:
I was able to compile and run this following [this tutorial](https://github.com/felixendres/rgbdslam_v2/wiki/Instructions-for-Compiling-Rgbdslam-(V2)-on-a-Fresh-Ubuntu-16.04-Install-(Ros-Kinetic)-in-Virtualbox), written by the author. 

I believe using ROS kinetic + Ubuntu 16.04 will invariably yield this error, since it links PCL 1.7 which breaks with C++11. This is weird considering it's the recommended system inside the readme.