felixendres/rgbdslam_v2 #107: error: invalid conversion from ‘int’ to ‘cv::ORB::ScoreType’

vahidbehtajisiahkal, 2019-01-17:
/home/vahid/catkin_realsense/src/rgbdslam_v2-kinetic/src/feature_adjuster.cpp: In member function ‘virtual void DetectorAdjuster::detect(cv::InputArray, std::vector<cv::KeyPoint>&, cv::InputArray)’:
/home/vahid/catkin_realsense/src/rgbdslam_v2-kinetic/src/feature_adjuster.cpp:94:87: error: invalid conversion from ‘int’ to ‘cv::ORB::ScoreType’ [-fpermissive]
       detector = ORB::create(10000, 1.2, 8, 15, 0, 2, 0, 31, static_cast<int>(thresh_));
                                                                                       ^
In file included from /home/vahid/catkin_realsense/src/rgbdslam_v2-kinetic/src/feature_adjuster.h:3:0,
                 from /home/vahid/catkin_realsense/src/rgbdslam_v2-kinetic/src/feature_adjuster.cpp:43:
/usr/local/include/opencv4/opencv2/features2d.hpp:346:29: note:   initializing argument 7 of ‘static cv::Ptr<cv::ORB> cv::ORB::create(int, float, int, int, int, int, cv::ORB::ScoreType, int, int)’
     CV_WRAP static Ptr<ORB> create(int nfeatures=500, float scaleFactor=1.2f, int nlevels=8, int edgeThreshold=31,
                             ^~~~~~
/home/vahid/catkin_realsense/src/rgbdslam_v2-kinetic/src/feature_adjuster.cpp:111:89: error: invalid conversion from ‘int’ to ‘cv::ORB::ScoreType’ [-fpermissive]
         detector = ORB::create(10000, 1.2, 8, 15, 0, 2, 0, 31, static_cast<int>(thresh_));

Jackwitt77, 2020-03-11:
I'm also getting this error.

Willforcv, 2021-11-17:
anyone solve this question？