felixendres/rgbdslam_v2 #120: Error while loading shared libraries: libg2o_ext_freeglut_minimal.so: cannot open shared object file: No such file or directory

ankurjay, 2020-07-27:
I am trying to launch the package while my gazebo simulation (that has some robots with kinects) in it. However, the following message appears and everything shuts down:

```
process[ankurbot_1/rgbdslam-8]: started with pid [12809]
/home/ankur/catkin_ws/devel/lib/rgbdslam/rgbdslam: error while loading shared libraries: libg2o_ext_freeglut_minimal.so: cannot open shared object file: No such file or directory
================================================================================REQUIRED process [ankurbot_1/rgbdslam-8] has died!
process has died [pid 12809, exit code 127, cmd /home/ankur/catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/ankur/.ros/log/0c84ebb4-d04e-11ea-9e9a-5435308b6f49/ankurbot_1-rgbdslam-8.log].
log file: /home/ankur/.ros/log/0c84ebb4-d04e-11ea-9e9a-5435308b6f49/ankurbot_1-rgbdslam-8*.log
Initiating shutdown!
================================================================================
```

If I launch the package without running my gazebo simulation, the error still persists. What can I do to resolve this?

felixendres, 2020-08-27:
It seems you are missing a library that should come with g2o. I do not know how to solve this. Do you have libg2o_ext_freeglut_minimal.so in your system?

Bardreamaster, 2021-05-19:
I have the same problem. 
But I have make sure that there's libg2o_ext_freeglut_minimal.so file in my system. By using
`ldconfig -v | grep "g2o"`
I can find it.
But it still report the error
~Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam: error while loading shared libraries: libg2o_ext_freeglut_minimal.so: cannot open shared object file: No such file or directory
================================================================================REQUIRED process [rgbdslam-2] has died!
process has died [pid 19169, exit code 127, cmd /home/changshanshi/Code/rgbdslam_catkin_ws/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/changshanshi/.ros/log/8c85926c-b8c1-11eb-afea-6807157e95b4/rgbdslam-2.log].
log file: /home/changshanshi/.ros/log/8c85926c-b8c1-11eb-afea-6807157e95b4/rgbdslam-2*.log
Initiating shutdown!
=====================

Bardreamaster, 2021-05-19:
I have solve it.
1. add the lib path where the lib you missed into /etc/ld.so.conf 
2. using `sudo ldconfig` to make it working
3. finally roslaunch

closed 2021-06-12 by felixendres