felixendres/rgbdslam_v2 #124: REQUIRED process [rgbdslam-25] has died!

fusionpdc, 2020-10-19:
process has died [pid 31950, exit code -11, cmd /home/jibaolin/catkin/catkin_ws_rgbd/devel/lib/rgbdslam/rgbdslam __name:=rgbdslam __log:=/home/jibaolin/.ros/log/3108c0e8-1200-11eb-a781-70c94ee92f41/rgbdslam-25.log].
log file: /home/jibaolin/.ros/log/3108c0e8-1200-11eb-a781-70c94ee92f41/rgbdslam-25*.log
Initiating shutdown!

How can I solve this problem.

felixendres, 2020-11-04:
You need to give more details. This text just says it crashed because of a segfault. I have no idea what you are doing, how you built rgbdslam and its dependency, on what operating system (with version) and so on.

HarryF6, 2021-02-06:
@fusionpdc did you solve it?

HarryF6, 2021-02-06:
@fusionpdc did you solve it? because i have the same problem, i am using a virtual machine of ubuntu 16.04 and ros kinetic to make it work

felixendres, 2023-07-18:
Remember this is a very generic problem description. The OP may have a very different root cause than anyone else who gets a segfault.

If you report such a problem, please include more information.

A good way to do this is either
- [enable core dumps](https://stackoverflow.com/questions/17965/how-to-generate-a-core-dump-in-linux-on-a-segmentation-fault), let rgbdslam_v2 crash, [open the core dump in gdb and print a backtrace](https://stackoverflow.com/questions/8806469/how-to-analyze-core-dump?rq=3)
- Launch rgbdslam_v2 in the debugger, by [adding a launch-prefix](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) in the launch file: `launch-prefix="gdb -ex run --args"`. Then, let it crash and type `bt` when the gdb prompt appears, to obtain a backtrace.

closed 2023-07-18 by felixendres